UVVM/UVVM #91: BUG while compiling lates UVVM with GHDL

ZirconfleX, 2019-11-27:
Description
- Latest version of GHDL installed on Linux-Mint_19.2(Tara).
- GHDL installed using LLVM approach. (Used LLVM version 9).
- GHDL installed in /opt/Ghdl (directory is owned by the )
- Downloaded the libraries from UVVM github page (UVVM version 2019.11.25).
- Downloaded zip extracted in /opt/Ghdl/lib/ghdl/UVVM
  - Subdirectory UVVM-2019.11.25 is created containing all UVVM stuff (files and directories)
- Adapted the GHDL config.sh file with these settings:
  - InstallationDirectories[UVVM]="/opt/Ghdl/lib/ghdl/UVVM/UVVM-2019.11.25"
  - DestinationDirectories[UVVM]="/opt/Ghdl/lib/ghdl/UVVM"
  - SourceDirectories[UVVM]="."
- In a terminal changed the working directory to: `cd opt/Ghdl/lib/`
- Execute here command: `./vendors/compile-uvvm.sh --uvvm-utilities`
This ends in a successful compile and a uvvm_util is created as sub-directory in /opt/Ghdl/lib/ghdl/UVVM

Run as second command: `./vendors/compile-uvvm.sh --uvvm-vvc-framework`
This ends in below BUG report, but the uvvm_vvc_framework directory is created.

Expected behaviour
I reported this as a GHDL BUG, but it does not seem it is!
I downloaded an earlier UVVM version (UVVM_2019-09-02) and there everything works perfect.
can do `./vendors/compile-uvvm --all` or `--uvvm` or `uvvm-vip` or ... everything ends successful.

Because of this, none of the other UVVM libraries can be compiled.
When executing: `./vendors/compile-uvvm.sh --uvvm` the *utilities* and *vvc-framework* libraries should be compiled, due to the BUG message and error *vvc-framework* is not created.
When executing: `./vendors/compile-uvvm.sh --uvvm-vip` all vip libraries should be created, but this does not happen because each of them need vvc-framework.

How to reproduce?
Do as described above in the description paragraph.

******************** GHDL Bug occurred ***************************
Please report this bug on https://github.com/ghdl/ghdl/issues
GHDL release: 0.37-dev (tarball) [Dunoon edition]
Compiled with GNAT Version: Community 2019 (20190517-83)
Target: x86_64-pc-linux-gnu
/opt/Ghdl/lib/ghdl/UVVM/uvvm_vvc_framework/v08/
Command line:
/opt/Ghdl/bin/ghdl1-llvm -fexplicit -frelaxed-rules --no-vital-checks --warn-binding --mb-comments --std=08 --work=uvvm_vvc_framework -P/opt/Ghdl/lib/ghdl/UVVM/ -P/opt/Ghdl/lib/ghdl/ieee/v08/ -P/opt/Ghdl/lib/ghdl/ -c -fpic -o ti_vvc_framework_support_pkg.o /opt/Ghdl/lib/ghdl/UVVM/UVVM-2019.11.25/uvvm_vvc_framework/script/../src/ti_vvc_framework_support_pkg.vhd
Exception PROGRAM_ERROR raised
Exception information:
raised PROGRAM_ERROR : ortho_llvm.adb:827 explicit raise
Call stack traceback locations:
0x414a0a 0x5687f0 0x56b8f0 0x56c632 0x56c994 0x56e8e5 0x56ff71 0x57150e 0x5b167a 0x5b1915 0x59bedd 0x60d869 0x60ec21 0x61000e 0x4084bd 0x7feb798a5b95 0x4076d8 0xfffffffffffffffe
******************************************************************

Additional context
Below is the terminal output after the bug message.

/opt/Ghdl/bin/ghdl: compilation error
Analyzing package '/opt/Ghdl/lib/ghdl/UVVM/UVVM-2019.11.25/uvvm_vvc_framework/script/../src/ti_generic_queue_pkg.vhd'
/opt/Ghdl/lib/ghdl/UVVM/UVVM-2019.11.25/uvvm_vvc_framework/script/../src/ti_generic_queue_pkg.vhd:69:16:warning: declaration of "scope" hides constant interface "scope" [-Whide]
/opt/Ghdl/lib/ghdl/UVVM/UVVM-2019.11.25/uvvm_vvc_framework/script/../src/ti_generic_queue_pkg.vhd:72:16:warning: declaration of "scope" hides constant interface "scope" [-Whide]
/opt/Ghdl/lib/ghdl/UVVM/UVVM-2019.11.25/uvvm_vvc_framework/script/../src/ti_generic_queue_pkg.vhd:550:14:warning: declaration of "scope" hides constant interface "scope" [-Whide]
/opt/Ghdl/lib/ghdl/UVVM/UVVM-2019.11.25/uvvm_vvc_framework/script/../src/ti_generic_queue_pkg.vhd:574:14:warning: declaration of "scope" hides constant interface "scope" [-Whide]
Analyzing package '/opt/Ghdl/lib/ghdl/UVVM/UVVM-2019.11.25/uvvm_vvc_framework/script/../src/ti_data_queue_pkg.vhd'
/opt/Ghdl/lib/ghdl/UVVM/UVVM-2019.11.25/uvvm_vvc_framework/script/../src/ti_data_queue_pkg.vhd:27:19:warning: type of a shared variable must be a protected type [-Wshared]
Analyzing package '/opt/Ghdl/lib/ghdl/UVVM/UVVM-2019.11.25/uvvm_vvc_framework/script/../src/ti_data_fifo_pkg.vhd'
Analyzing package '/opt/Ghdl/lib/ghdl/UVVM/UVVM-2019.11.25/uvvm_vvc_framework/script/../src/ti_data_stack_pkg.vhd'
Analyzing package '/opt/Ghdl/lib/ghdl/UVVM/UVVM-2019.11.25/uvvm_vvc_framework/script/../src/ti_uvvm_engine.vhd'
/opt/Ghdl/lib/ghdl/UVVM/UVVM-2019.11.25/uvvm_vvc_framework/script/../src/ti_uvvm_engine.vhd:24:24: unit "ti_vvc_framework_support_pkg" not found in library "uvvm_vvc_framework"
/opt/Ghdl/lib/ghdl/UVVM/UVVM-2019.11.25/uvvm_vvc_framework/script/../src/ti_uvvm_engine.vhd:29:14: entity 'ti_uvvm_engine' was not analysed
/opt/Ghdl/bin/ghdl: compilation error
--------------------------------------------------------------------------------
Compiling UVVM packages [FAILED]

UVVM, 2019-11-28:
Hi,
This sound a bit strange as not much have change from release v2019.09.02 to v2019.11.25 in uvvm_vvc_framework. I have downloaded latest GHDL release (compiled with mcode) and UVVM release v2019.11.25, and compiling uvvm_util, followed by uvvm_vvc_framework is not giving me any problems. 
The VIPs will not compile until GHDL update their compile scripts to make sure that bitivis_vip_scoreboard is compiled prior to any other VIP. In the  VVC QRs compile order section you see that we have included scoreboard to all VIPs, thus the VIPs depend on library bitvis_vip_scoreboard to be compiled. A temporarily fix can be to comment out all scoreboard code in the VIPs (vvc_methods_pkg.vhd and <interface>_vvc.vhd).

Best regards
Marius

ZirconfleX, 2019-11-28:
Hallo Marius,

I'm using GHDL compiled with LLVM.
     Find a chapter, about installing GHDL, I extracted from a document 
I wrote on how to install Linux-Mint_19 on a PC.
     I'm using GHDL like that already for a while, compiled Xilinx and 
Lattice libraries, and all works perfect.

I've tried it a couple of times,
     V2019.11.25 has issues with the compilation of uvvm-vvc-framework 
and due to that compiling ViP goes wrong too.
     With V2019.09.02 compilation runs as a train until the end 
(Compiling UVVM packages [SUCCESSFUL]).
     I downloaded now from https://github.com/UVVM/UVVM/releases version 
2019.11.21
         Compiling of uvvm-utilities end with:     Compiling UVVM 
packages [SUCCESSFUL]
         Compiling of uvvm-vvc-framework ends with a BUG output, view 
the attached .txt file.
             A uvvm-vvc-framwork directory is created and it contains a 
sub-directory v08 with files.
             I zipped that directory and dropped it in the attached zip.

When I return to version 2019.09.02 and perform the same operations, then
     Compiling of uvvm-utilities end with:    Compiling UVVM packages 
[SUCCESSFUL]
     Compiling of uvvm-vvc-framework ends with:    Compiling UVVM 
packages [SUCCESSFUL]
     Now I can also do a successful compile of all VIP libraries.

finally:
     V2019.11.21 and V2019.11.25 both error out with a BUG message, when 
compiling uvvm-vvc-framework
     V2019.09.02 compiles perfectly.

Hope it can shine a bit more light on the case.

Kind regards,

Marc

mavric315, 2019-12-18:
I have also found that GHDL is no longer able to compile UVVM. I traced it back to UVVM 2019.09.02 works but the next release, 2019.11.21 and all subsequent releases fail to build.  I am running the powershell build scripts and GHDL version 0.36 llvm. The only change is which UVVM version I am compiling.  The GHDL compile-uvvm.ps1 script does have an error in that it does not like the '#' comment line in the beginning of the compile_order.txt files but build does continue unaffected.  I have attached two log files. The build that works against 2019.09.02 and the latest release that fails 2019.12.04. The first error encountered is:

ERROR: D:\fpgaTools\UVVM-master\uvvm_vvc_framework\src\ti_vvc_framework_support_pkg.vhd:26:24: unit "ti_protected_types_pkg" not found in library "uvvm_vvc_framework"

-Don

[build-2019.09.02-ok.txt](https://github.com/UVVM/UVVM/files/3980968/build-2019.09.02-ok.txt)
[build-2019.12.04-error.txt](https://github.com/UVVM/UVVM/files/3980963/build-2019.12.04-error.txt)

UVVM, 2019-12-19:
Hi,
The error message claims the ti_protected_types_pkg is not found in the uvvm_vvc_framework library. This package is required by the activity watchdog functionality, which all VVCs are registered in, and was introduced in release v2019.11.21. and has to  be compiled prior to ti_vvc_framework_support_pkg as it has dependency to the ti_protected_types_pkg.

The compile order is given in the script/compile_order.txt file :
ti_protected_types_pkg.vhd
ti_vvc_framework_support_pkg.vhd
ti_generic_queue_pkg.vhd
ti_data_queue_pkg.vhd
ti_data_fifo_pkg.vhd
ti_data_stack_pkg.vhd
ti_uvvm_engine.vhd

Best regards
Marius

mavric315, 2019-12-19:
Hi Marius,

Thanks for the pointer. As it turns out 0.36 of GHDL does not use the compile_order.txt file for building the uvvm_util or the uvvm_vvc_framework in the powershell and bash scripts. The files and compile order are hard coded in the compile_uvvm.ps1 / compile_uvvm.sh scripts.  And as far as I can tell only the powershell script uses the compile_order.txt file for all the VVCs. It looks like the bash script has the files hard coded. I updated locally the GHDL script to include ti_protected_types_pkg.vhd and was able to build but then ran into the "GHDL Bug Occurred" issue reported at the top of this thread.  So I am reverting to 2019.09.02 for now.

Thanks again for the help!

-Don

GlenNicholls, 2020-07-14:
@ZirconfleX If you are still having this problem, please try with either GHDL0.37 or one of the nightly builds found [here](https://github.com/ghdl/ghdl/releases). If you're still seeing that problem, open an issue on the GHDL GitHub with the information. I see @diogratia opened an issue with GHDL that has been closed, so I'm curious as to why this is still open.

GlenNicholls, 2020-07-14:
@mavric315 I am looking at this line https://github.com/ghdl/ghdl/blob/99b542c849311c92e87e2c70d283de133c9d4093/libraries/vendors/compile-uvvm.sh#L302

It looks like it is using `compile_order.txt`, so now I'm not sure why this issue is still open. Has anyone tried recently to compile UVVM with the latest release of GHDL?

ZirconfleX, 2020-07-15:
Hi,

Tried the whole trajectory.

Installed the latest GNAT tool chain.
Installed the nearly newest LLVM  (newest is version 10, but I used
version 9)
Went successfully through the GHDL installation process.
Went successfully trough the compilation process for Xilinx Vivado.
Downloaded the latest UVVM-master
Went successfully through teh compilation process.

BUT:

Then I saw that the latest UVVM-master directory contained more
components then compiled using the GHDL compile_uvvm.sh script.
Went looking in the UVVM doc and found a description pointing to a
"compile_order.txt" file.
That file is not available in the GHDL vendors scripts directory.
There is a "compile_order.txt" file in the UVVM/script directory but I
don't know how to use it.
 
Opened the GHDL compile_uvvm.sh file and surprise, nearly at the top is
a list of modules to compile similar as in the "compile_order.txt" file.

- Add to uvvm_pkgs:
uvvm_util
uvvm_vvc_framework
xConstrRandFuncCov

 - Add to uvvm_vips
bitvis_vip_scoreboard
bitvis_vip_avalon_mm
bitvis_vip_avalon_st
bitvis_vip_axilite
bitvis_vip_axistream
bitvis_vip_clock_generator
bitvis_vip_error_injection
bitvis_vip_gmii
bitvis_vip_gpio
bitvis_vip_i2c
bitvis_vip_rgmii
bitvis_vip_sbi
bitvis_vip_spi
bitvis_vip_uart
bitvis_vip_hvvc_to_vvc_bridge
bitvis_vip_ethernet
bitvis_irqc
bitvis_vip_spec_cov

Running the compilation seems to do fine but it ends with:   Compiling
UVVM packages [FAILED]
Looking into the directory with the created simulation models seems that
everything is there.
Browsing up through the compiler output text in the terminal learns that
for one module it went wrong.

Compiling library 'bitvis_vip_hvvc_to_vvc_bridge'...
Analyzing package
'/opt/Ghdl/lib/ghdl/UVVM-master/bitvis_vip_hvvc_to_vvc_bridge/script/../src/support_pkg.vhd'

******************** GHDL Bug occurred ***************************
Please report this bug on https://github.com/ghdl/ghdl/issues
GHDL release: 1.0-dev () [Dunoon edition]
Compiled with GNAT Version: Community 2020 (20200429-93)
Target: x86_64-pc-linux-gnu
/opt/Ghdl/lib/ghdl/UVVM/bitvis_vip_hvvc_to_vvc_bridge/v08/
Command line:
/opt/Ghdl/bin/ghdl1-llvm --std=08 -fexplicit -frelaxed-rules
--no-vital-checks --warn-binding --mb-comments
--work=bitvis_vip_hvvc_to_vvc_bridge -P/opt/Ghdl/lib/ghdl/UVVM/
-P/opt/Ghdl/lib/ghdl/ieee/v08/ -P/opt/Ghdl/lib/ghdl/ -c -fpic -o
support_pkg.o
/opt/Ghdl/lib/ghdl/UVVM-master/bitvis_vip_hvvc_to_vvc_bridge/script/../src/support_pkg.vhd
Exception TYPES.INTERNAL_ERROR raised
Exception information:
raised TYPES.INTERNAL_ERROR : trans-chap3.adb:2967
Call stack traceback locations:
0x57c813 0x56c597 0x5f66f4 0x60eeb0 0x60f8e1 0x611aae 0x611c17 0x5dc9c1
0x5f29aa 0x5ddb82 0x5c7e15 0x625909 0x626ce7 0x628048 0x408c49
0x7f04d30a00b1 0x407e2c 0xfffffffffffffffe
******************************************************************
/opt/Ghdl/bin/ghdl: compilation error

All modules were this "bitvis_vip_hvvc_to_vvc_bridge" is used also fil
to compile correctly (obvious)
Those are:
    gmii
    sbi
    ethernet.

I've written the whole process down and it will appear here:
https://zirconflex.github.io/DeKamiel/

Kind regards,

Marc

ZirconfleX, 2020-07-15:
Edit the list in the "compile_uvvm.sh" GHDL vendors script.
Add as earlier stated:
- Add to uvvm_pkgs:
uvvm_util
uvvm_vvc_framework
xConstrRandFuncCov

 - Add to uvvm_vips
bitvis_vip_scoreboard
bitvis_vip_avalon_mm
bitvis_vip_avalon_st
bitvis_vip_axilite
bitvis_vip_axistream
bitvis_vip_clock_generator
bitvis_vip_error_injection
bitvis_vip_gmii
bitvis_vip_gpio
bitvis_vip_i2c
bitvis_vip_rgmii
bitvis_vip_sbi
bitvis_vip_spi
bitvis_vip_uart
bitvis_irqc
bitvis_vip_spec_cov

But  leave out:
bitvis_vip_hvvc_to_vvc_bridge
bitvis_vip_ethernet

Then a compile ends: Compiling UVVM packages [SUCCESSFUL]

The error is only present in the "bitvis_vip_hvvc_to_vvc_bridge" code
    This code uses parts of "bitvis_vip_gmii" and "bitvis_vip_sbi" .
    It reports these models in the error output and by that it seems
that those models are wrong too.

Because the error appears in "bitvis_vip_hvvc_to_vvc_bridge" and
"bitvis_vip_ethernet" uses that core, it errors out too.

Regards,

Marc

GlenNicholls, 2020-07-15:
@ZirconfleX Okay, thanks for running that! I will open an issue on the GHDL repo and link to this issue. Can you run `ghdl --version` and provide the commit tag?

e.g.:

```
$ ghdl --version
GHDL 1.0-dev (v0.37.0-760-g**1666a4c7**) [Dunoon edition]
 Compiled with GNAT Version: 9.1.0
 llvm code generator
Written by Tristan Gingold.

Copyright (C) 2003 - 2020 Tristan Gingold.
GHDL is free software, covered by the GNU General Public License.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE
```

ZirconfleX, 2020-07-16:
> e.g., in my case it is 1666a4c7

Here you go:

marc@Asusium:~$ ghdl --version
GHDL 1.0-dev () [Dunoon edition]
 Compiled with GNAT Version: Community 2020 (20200429-93)
 llvm code generator
Written by Tristan Gingold.

Copyright (C) 2003 - 2020 Tristan Gingold.
GHDL is free software, covered by the GNU General Public License.  There
is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

Regards,

ZirconfleX, 2020-07-16:
> @ZirconfleX <https://github.com/ZirconfleX> Alright, this issue is now
> gated by the GHDL issue here ghdl/ghdl#1393

I've yesterday posted the article how to do the things for GHDL and
GtkWave on : https://zirconflex.github.io/DeKamiel/
under articles: Install GHDL and GtkWave on Linux-Mint or Ubuntu

tpcorrea, 2021-01-17:
Hi all! The scripts have been updated in the GHDL repository, but there are still some VIPs that fails to compile. Please check the new scripts.

UVVM, 2021-01-17:
Hi,
What are the error messages?

Br,
Marius

tpcorrea, 2021-01-17:
VIP_AVALON_ST:
[image]

VIP_ETHERNET:
[image]

tpcorrea, 2021-01-17:
The first had a GHDL exception, but the VIP_ETHERNET had errors on its own.

Paebbels, 2021-01-17:
For compatibility problems between GHDL and UVVM see: #130 and ghdl/ghdl#1593

m-kru, 2021-05-18:
Current master of GHDL compiles current master of UVVM correctly.

closed 2021-05-26 by mariuselv