UVVM/UVVM #93: Using VVC in function decleared outside of tb.vhd

cartfjord, 2019-12-06:
Hi, I have a function decleared in an external package where I want to use the SPI VVC. However, I am not able to pass the SPI_VVCT to my function since t_vvc_target_record is decleared in td_target_support_pkg.vhd.

I have tried including the td_target_support_pkg as follows, but no success:

library uvvm_vvc_framework;
use uvvm_vvc_framework.ti_vvc_framework_support_pkg.all;
use uvvm_vvc_framework.td_target_support_pkg.all;

 Any suggestions?

UVVM, 2019-12-09:
Hi, 
have you included the VVC library, e.g. bitvis_vip_sbi in your package? 
`library bitvis_vip_sbi;
context bitvis_vip_sbi.vvc_context;
use bitvis_vip_sbi.sbi_bfm_pkg.all;`

Best regards
Marius

cartfjord, 2019-12-11:
I am sorry for my late response. I have just checked now. It still does not work. My procedure declaration looks as follows:

package body my_pkg is

procedure foo(
		signal VVCT     : inout t_vvc_target_record;
                signal a            : std_logic;
                signal b            : std_logic
) is ..........

I get the following error message:

"
Multiple markers at this line
- Could not find declaration corresponding to 't_vvc_target_record'.
- Unused declaration
- Vcom/Vlog: (vcom-1136) Unknown identifier "t_vvc_target_record
"

Trying to include a 

"use work.td_target_support_pkg.all;" 

It complains that it does not exist:

"
Multiple markers at this line
- Could not find declaration corresponding to 'td_target_support_pkg'.
- Vcom/Vlog: Unknown expanded name.
- Vcom/Vlog: (vcom-1195) Cannot find expanded name "work.td_target_support_pkg".
"

Maybe my compiler order is somehow wrong? I use Sigasi if that matters.

UVVM, 2019-12-13:
Is package my_pkg compiled to the same library as the VVC? The file td_target_support_pkg.vhd is compiled in to the VVC library. The correct compile order is listed in the VVCs quick reference on the last page, and also in the /script folder of the VVCs. 
If you like we could have a look at you setup, just send it on info[at]bitivis.no

Best regards,
Marius

closed 2020-01-08 by UVVM