UVVM/UVVM #94: Missing feature to deallocate UVVM FIFO

fergzie, 2019-12-11:
I'd like to request a feature that I'm missing. There is no way to deallocate a UVVM FIFO, but the data queue package, which UVVM FIFO is built upon, does support deallocation of buffer.

UVVM, 2020-01-07:
Hi,

Yes there is a procedure deallocate_buffer() which resets all the pointers in the buffer thus deallocating ALL the queues. Is this the feature you would like for the UVVM FIFO? Because deallocating a single FIFO requires a more complex procedure.

Best regards,
Erick

fergzie, 2020-01-07:
Hi Erick,

Yes I'd like that feature for the UVVM FIFO.

Br,
David

UVVM, 2020-01-07:
Alright, then we'll make sure to add that feature in our next release.

fergzie, 2020-01-07:
Great. Thank you :+1:

UVVM, 2020-01-24:
New release is out.

closed 2020-01-24 by UVVM