UVVM/UVVM #95: Continuous output of VVC

olagrottvik, 2019-12-16:
Hi,

I'm trying to create a VVC for 10GBASE-R. However, it requires a continuous tx-output (and continuous parsing of rx-input) for the encoding to always be correct. I would think this would be possible if one could check the VVC queue for commands, and if the queue was empty, it would run a procedure that made sure some output was transmitted. However, now the VVC is locked in await_cmd_from_sequencer. Is there some way to work around this problem? I've seen it done for Verilog using Icarus and a myhdl python script but it would be nice to have all testbenches in VHDL + UVVM. 

Example python: https://github.com/alexforencich/verilog-
ethernet/blob/master/tb/baser_serdes_ep.py

UVVM, 2019-12-17:
Hi Ola,
This should be possible to implement, although it will require that you do some modifications:
You will need a loop which continuously checks if there is a command in the queue, and if not it will call the TX method with the type of idle data you want to send (I'm guessing this is some sort of recognizable pattern). 

Best regards,
Marius

UVVM, 2020-01-24:
Closing.
The question has been moved to the UVVM forum: https://forum.uvvm.org/t/continuous-output-of-vvc/24

closed 2020-01-24 by UVVM