UVVM/UVVM #96: axilite_bfm_pkg t_axilite_bfm_config num_X_pipe_stages

pkinzer, 2019-12-16:
The num_r_pipe_stages and num_b_pipe_stages in t_axilite_bfm_config does not appear to be used.   Bus cycle length is not affected.  What would it take to get support for these config values added?

Thanks,
-Paul

UVVM, 2019-12-17:
Hi Paul,
I have created an internal ticket for your request. Unfortunately there are few available resources at the moment, but we are continuously adding new features to our VIPs. I will update your request as soon as we are ready to start the implementation.

Best regards,
Marius

pkinzer, 2020-01-07:
I've added pull request 97 to address this.

UVVM, 2020-01-17:
Hi Paul,
This could be a very useful pull request, and we will handle it as soon as possible. 
At the moment we are in the midst of making a new release with some important changes on BFM and VVC interface setup plus some critical changes to reduce the memory usage when using the new Direct Transaction Transfer. We hope to do this release mid/end next week and will look at your pull request immediately after that.

Best regards
Espen

UVVM, 2020-01-22:
Hi Paul,

We tested your changes with our internal testbenches and there seems to be a problem. As mentioned before we don't have time at the moment to check why it doesn't work so perhaps you could use our testbenches and try them out yourself?
You can send an email to support@bitvis.no so we can send you the files.

Best regards,
Erick

pkinzer, 2020-07-21:
Updated patch-1 to UVVM 2020-06-23, tested with my internal AXI4 master & slave. Still fails axilite_simple_tb.vhd, which appears to be an AMBA AXI Protocol Specification violation in axilite_slave.vhd and/or axilite_slave_model.vhd.

Error: "axilite_write(A:x"0000", x"5555") => Failed. Boolean was false. ': Timeout waiting for BVALID"

Observing the bus cycles, axilite_if_2 is asserting write_response_channel.bvalid, then de-asserting before write_response_channel.bready is asserted. 

The AMBA AXI Protocol Specfication, section A3.2.2, states "The slave can assert BVALID signal only when it drives a valid write response.  When asserted, BVALID must remain asserted until the rising clock edge after the master asserts BREADY"

UVVM, 2020-08-24:
Hi Paul,

We have fixed our internal testbenches and incorporated your pull request.
Thank you for your contribution and sorry for the delay.

Best regards,
Erick

closed 2020-08-24 by UVVM

pkinzer, 2020-08-24:
Thanks Erick.