UVVM/UVVM #98: entity should end with end <entity_name>; and not end entity;

chevillotn, 2020-01-17:
There are multiples files with the following:
```
entity entity_name is
...
end entity;
```
Usually the entity should end with end entity_name;
See
https://www.ics.uci.edu/~jmoorkan/vhdlref/entity.html

UVVM, 2020-01-20:
Hi,
Our interpretation of the entity declaration syntax rule is that the declaration completion can consist of the end keyword only, or along with any combination of the entity keyword and identifier.
A quick search in the current release of UVVM there are mostly demo test benches that have an entity declaration completion without any identifier. Although not strictly necessary, we can changes this in those files.

Best regards,
Marius

closed 2020-01-20 by UVVM

chevillotn, 2020-01-20:
Hi Marius,
I was actually trying to find the VHDL specification to try to see exactly what should be. It works in both cases of course.
Would you know where to find this spec?

chevillotn, 2020-01-20:
I see you have closed it, so i understand you will keep the code as it is?

UVVM, 2020-01-21:
Hi,
No, we are making the changes and it will be part of our next release.

Br, 
Marius

chevillotn, 2020-01-21:
Hi Marius,
I appreciate but then i don't understand the workflow. When you work on an issue you close the issue? This is mostly confusing.