UVVM/UVVM #99: Unable to disable_log_msg

fransschreuder, 2020-02-12:
Hello,
I am happily using UVVM, however I have a lot of check_value calls in my testbench. For everyone I get a log message if the check is successful.
I tried calling:
`disable_log_msg(ID_POS_ACK);`
However the messages keep occurring. Am I missing something?
Thanks for the help.

UVVM, 2020-02-12:
Hi Frans, and great to hear that you enjoy using UVVM!
It should be sufficient to us e disable_log_msg(ID_POS_ACK). Are the checks located in the test sequencer?

Best regards,
Marius

fransschreuder, 2020-02-12:
Yes, the tests are in the sequencer.

UVVM, 2020-02-12:
Could there be any checks that is reportiing from a VVC or BFM? Can you try to disable all messages, e.g. disable_log_msg(VVC_BROADCAST, ALL_MESSAGES) ?

fransschreuder, 2020-02-12:
There is no VCC or BFM in my testbench, just the UUT and a sequencer process. VVC_BROADCAST is undefined (maybe that's a clue?).
I am still fairly new to UVVM and so far mainly using the uvvm_utils.

UVVM, 2020-02-12:
I see. Can you paste the test output?

fransschreuder, 2020-02-12:
Here is the important part of the output. Only included a few of the logs, it keeps on going after this...
```
# UVVM:      
# UVVM:      --------------------------------------------------------------------------------------------------------------------------------------------------------------------
# UVVM:      ***  REPORT OF GLOBAL CTRL ***
# UVVM:      --------------------------------------------------------------------------------------------------------------------------------------------------------------------
# UVVM:                                IGNORE    STOP_LIMIT
# UVVM:                NOTE         :  REGARD         0
# UVVM:                TB_NOTE      :  REGARD         0
# UVVM:                WARNING      :  REGARD         0
# UVVM:                TB_WARNING   :  REGARD         0
# UVVM:                MANUAL_CHECK :  REGARD         0
# UVVM:                ERROR        :  REGARD         1
# UVVM:                TB_ERROR     :  REGARD         1
# UVVM:                FAILURE      :  REGARD         1
# UVVM:                TB_FAILURE   :  REGARD         1
# UVVM:      --------------------------------------------------------------------------------------------------------------------------------------------------------------------
# UVVM:      
# UVVM:      
# UVVM:      --------------------------------------------------------------------------------------------------------------------------------------------------------------------
# UVVM:      ***  REPORT OF MSG ID PANEL ***
# UVVM:      --------------------------------------------------------------------------------------------------------------------------------------------------------------------
# UVVM:                ID                             Status
# UVVM:                ------------------------       ------
# UVVM:                ID_UTIL_BURIED               : DISABLED
# UVVM:                ID_BITVIS_DEBUG              : DISABLED
# UVVM:                ID_UTIL_SETUP                : ENABLED
# UVVM:                ID_LOG_MSG_CTRL              : ENABLED
# UVVM:                ID_ALERT_CTRL                : ENABLED
# UVVM:                ID_FINISH_OR_STOP            : ENABLED
# UVVM:                ID_CLOCK_GEN                 : ENABLED
# UVVM:                ID_GEN_PULSE                 : ENABLED
# UVVM:                ID_BLOCKING                  : ENABLED
# UVVM:                ID_WATCHDOG                  : ENABLED
# UVVM:                ID_POS_ACK                   : ENABLED
# UVVM:                ID_LOG_HDR                   : ENABLED
# UVVM:                ID_LOG_HDR_LARGE             : ENABLED
# UVVM:                ID_LOG_HDR_XL                : ENABLED
# UVVM:                ID_SEQUENCER                 : ENABLED
# UVVM:                ID_SEQUENCER_SUB             : ENABLED
# UVVM:                ID_BFM                       : ENABLED
# UVVM:                ID_BFM_WAIT                  : ENABLED
# UVVM:                ID_BFM_POLL                  : ENABLED
# UVVM:                ID_BFM_POLL_SUMMARY          : ENABLED
# UVVM:                ID_TERMINATE_CMD             : ENABLED
# UVVM:                ID_SEGMENT_INITIATE          : ENABLED
# UVVM:                ID_SEGMENT_COMPLETE          : ENABLED
# UVVM:                ID_SEGMENT_HDR               : ENABLED
# UVVM:                ID_SEGMENT_DATA              : ENABLED
# UVVM:                ID_PACKET_INITIATE           : ENABLED
# UVVM:                ID_PACKET_COMPLETE           : ENABLED
# UVVM:                ID_PACKET_HDR                : ENABLED
# UVVM:                ID_PACKET_DATA               : ENABLED
# UVVM:                ID_FRAME_INITIATE            : ENABLED
# UVVM:                ID_FRAME_COMPLETE            : ENABLED
# UVVM:                ID_FRAME_HDR                 : ENABLED
# UVVM:                ID_FRAME_DATA                : ENABLED
# UVVM:                ID_COVERAGE_MAKEBIN          : DISABLED
# UVVM:                ID_COVERAGE_ADDBIN           : DISABLED
# UVVM:                ID_COVERAGE_ICOVER           : DISABLED
# UVVM:                ID_COVERAGE_CONFIG           : ENABLED
# UVVM:                ID_COVERAGE_SUMMARY          : ENABLED
# UVVM:                ID_COVERAGE_HOLES            : ENABLED
# UVVM:                ID_UVVM_SEND_CMD             : ENABLED
# UVVM:                ID_UVVM_CMD_ACK              : ENABLED
# UVVM:                ID_UVVM_CMD_RESULT           : ENABLED
# UVVM:                ID_CMD_INTERPRETER           : ENABLED
# UVVM:                ID_CMD_INTERPRETER_WAIT      : ENABLED
# UVVM:                ID_IMMEDIATE_CMD             : ENABLED
# UVVM:                ID_IMMEDIATE_CMD_WAIT        : ENABLED
[... 120 lines cut ...]
# UVVM: ID_LOG_MSG_CTRL                  250.4 ns  TB seq.                        enable_log_msg(ID_UVVM_SEND_CMD). 
# UVVM: 
# UVVM: 
# UVVM: ID_LOG_HDR                       250.4 ns  GBTLinkToHost_tb               Starting simulation of TB for EGROUP using VVCs
# UVVM: -------------------------------------------------------------------------------------------------------------------------------------------------------------------------
# UVVM: ID_SEQUENCER                     250.4 ns  TB seq.                        Wait 10 clock period for reset to be turned off
# UVVM: ID_CLOCK_GEN                     250.4 ns  TB seq.                        Starting clock 250 MHz clock
# UVVM: ID_CLOCK_GEN                     250.4 ns  TB seq.                        Starting clock 240 MHz clock
[... cut ...]
# UVVM: ID_POS_ACK                    152926.4 ns  GBTLinkToHost_tb               await_value(slv x"1", 0 ns, 3000000 ns) => OK. Condition occurred after 10332 ns. 'wait for
# UVVM:                                                                           block to be written'
# UVVM: ID_POS_ACK                    153966.4 ns  GBTLinkToHost_tb               await_value(slv x"1", 0 ns, 3000000 ns) => OK. Condition occurred after 908 ns. 'wait for block
# UVVM:                                                                           to be written'
```

fransschreuder, 2020-02-12:
Oh never mind, there was another enable_log_message somewhere else in my code. Anyway, thanks for the quick support!

closed 2020-02-12 by fransschreuder