UVVM/UVVM #101: Avalon ST Receive for 16-bit and 32-bit data bus

carmen1403, 2020-03-11:
Hello,

I was usign the Avalon ST Bus VVC that you provide in your GiHub but it doesn't works for 16-bit and 32-bit data. I resolved this making some changes in the avalon_st_bfm_pkg.vhd file. Nevertheless, I think that there are another error in another file because looking the simulation it lost 2 packets in the receive process. Maybe you can take a look at this and verify the proper functionality for this type of data. 

  ---------------------------------------------------------------------------------------------
  -- Avalon-ST Receive
  -- DUT -> BFM
  ---------------------------------------------------------------------------------------------
  procedure avalon_st_receive (
    variable channel_value    : out   std_logic_vector;
    variable data_array       : out   t_slv_array;
    constant msg              : in    string                 := "";
    signal   clk              : in    std_logic;
    signal   avalon_st_if     : inout t_avalon_st_if;
    constant scope            : in    string                 := C_SCOPE;
    constant msg_id_panel     : in    t_msg_id_panel         := shared_msg_id_panel;
    constant config           : in    t_avalon_st_bfm_config := C_AVALON_ST_BFM_CONFIG_DEFAULT;
    constant ext_proc_call    : in    string := ""  -- External proc_call. Overwrite if called from other BFM procedure
  ) is  
    constant local_proc_name : string := "avalon_st_receive";  -- Internal proc_name; Used if called from sequencer or VVC
    constant local_proc_call : string := local_proc_name & "(" & to_string(data_array'length) & " sym)";
    constant c_data_word_size   : natural := data_array(data_array'low)'length;
    constant c_sym_width        : natural := config.symbol_width;
    constant c_symbols_per_beat : natural := avalon_st_if.data'length/config.symbol_width; -- Number of symbols transferred per cycle
    -- Normalize to the DUT channel/data widths
    variable v_normalized_chan : std_logic_vector(channel_value'length-1 downto 0) := (others => '0');
    variable v_normalized_data : t_slv_array(0 to data_array'length-1)(c_data_word_size-1 downto 0);
    -- Helper variables
    variable v_proc_call            : line; -- Current proc_call, external or local
    variable v_symbol_array         : t_slv_array_ptr;
    variable v_sym_in_beat          : natural := 0;
    variable v_sym_cnt              : natural := 0;
    variable v_invalid_count        : natural := 0;  -- # cycles without valid being asserted
    variable v_done                 : boolean := false;
    variable v_timeout              : boolean := false;
    variable v_empty_symbols        : natural := 0;
    variable v_data_offset          : natural := 0;
    variable v_time_of_rising_edge  : time    := -1 ns;  -- time stamp for clk period checking
    variable v_time_of_falling_edge : time    := -1 ns;  -- time stamp for clk period checking
  begin

    -- If called from sequencer/VVC, show 'avalon_st_receive()...' in log
    if ext_proc_call = "" then
      write(v_proc_call, local_proc_call);
    -- If called from other BFM procedure like avalon_st_expect, log 'avalon_st_expect() while executing avalon_st_receive()...'
    else
      write(v_proc_call, ext_proc_call & " while executing " & local_proc_name);
    end if;

    check_value(c_sym_width <= C_MAX_BITS_PER_SYMBOL, TB_FAILURE, "Sanity check: Check that symbol_width doesn't exceed C_MAX_BITS_PER_SYMBOL.", scope, ID_NEVER, msg_id_panel, v_proc_call.all);
    check_value(c_symbols_per_beat <= C_MAX_SYMBOLS_PER_BEAT, TB_FAILURE, "Sanity check: Check that c_symbols_per_beat doesn't exceed C_MAX_SYMBOLS_PER_BEAT.", scope, ID_NEVER, msg_id_panel, v_proc_call.all);
    check_value(avalon_st_if.data'length mod c_sym_width = 0, TB_FAILURE, "Sanity check: Check that data width is a multiple of symbol_width.", scope, ID_NEVER, msg_id_panel, v_proc_call.all);
    
    --alert(note, "symbols_per_beat: " & to_string(c_symbols_per_beat) & " empty_length: " & to_string(avalon_st_if.empty'length) & " c_sym_width: " & to_string(c_sym_width), scope);
    
    --check_value(avalon_st_if.empty'length = maximum(log2(c_symbols_per_beat),1), TB_FAILURE, "Sanity check: Check that empty width equals log2(symbols_per_beat).", scope, ID_NEVER, msg_id_panel, v_proc_call.all);
    
    check_value((c_data_word_size = c_sym_width) or (c_data_word_size = avalon_st_if.data'length), TB_FAILURE, "Sanity check: Check that data_array elements have either the size of the data bus or the configured symbol.", scope, ID_NEVER, msg_id_panel, v_proc_call.all);
    check_value(data_array'ascending, TB_FAILURE, "Sanity check: Check that data_array is ascending (defined with 'to'), for symbol order clarity.", scope, ID_NEVER, msg_id_panel, v_proc_call.all);
    if config.bfm_sync = SYNC_WITH_SETUP_AND_HOLD then
      check_value(config.clock_period > -1 ns, TB_FAILURE, "Sanity check: Check that clock_period is set.", scope, ID_NEVER, msg_id_panel, v_proc_call.all);
      check_value(config.setup_time < config.clock_period/2, TB_FAILURE, "Sanity check: Check that setup_time do not exceed clock_period/2.", scope, ID_NEVER, msg_id_panel, v_proc_call.all);
      check_value(config.hold_time < config.clock_period/2, TB_FAILURE, "Sanity check: Check that hold_time do not exceed clock_period/2.", scope, ID_NEVER, msg_id_panel, v_proc_call.all);
    end if;

    -- Use a symbol array to make it easier to iterate through the data
    if c_data_word_size = c_sym_width then
      v_symbol_array := new t_slv_array(0 to v_normalized_data'length-1)(c_sym_width-1 downto 0);
    else
      v_symbol_array := new t_slv_array(0 to v_normalized_data'length*c_symbols_per_beat-1)(c_sym_width-1 downto 0);
[... 131 lines cut ...]
          end if;
          v_normalized_data(i)(v_data_offset+c_sym_width-1 downto v_data_offset) := v_symbol_array(i*c_symbols_per_beat+j);
        end loop;
      end loop;
    end if;
    data_array    := v_normalized_data;
    check_value(to_integer(unsigned(v_normalized_chan)) <= config.max_channel, TB_FAILURE,
      "Sanity check: Check that channel number is supported.", scope, ID_NEVER, msg_id_panel, v_proc_call.all);
    channel_value := v_normalized_chan;

    avalon_st_if <= init_avalon_st_if_signals(is_master        => false,
                                              channel_width    => avalon_st_if.channel'length,
                                              data_width       => avalon_st_if.data'length,
                                              data_error_width => avalon_st_if.data_error'length,
                                              empty_width      => avalon_st_if.empty'length);

    -- Done. Check if there was a timeout or it was successful
    if v_timeout then
      alert(config.max_wait_cycles_severity, v_proc_call.all & "=> Failed. Timeout while waiting for valid data. " &
        add_msg_delimiter(msg) & "(c_sym_width): " & to_string(c_sym_width) & " (c_data_word_size):" & to_string(c_data_word_size) & 
        " (c_symbols_per_beat): " & to_string(c_symbols_per_beat), scope);
    else
      log(ID_PACKET_COMPLETE, v_proc_call.all & " DONE. " & add_msg_delimiter(msg), scope, msg_id_panel);
    end if;
  end procedure;

UVVM, 2020-03-11:
Hi,

We'll look into it as soon as possible.

Best regards,
Erick

carmen1403, 2020-03-11:
Hello,

Looking the simulation I was thinking that some packets are lost but now I stored the input data and the output data and comparing these two files the data it’s the same, so now with these changes it’s working.
Please verify this changes and this information and upload to GitHub because maybe another people in the future will have the same problem as me.

Thank you so much.
Best regards,
Carmen Misa

UVVM, 2020-03-11:
Hi,

Could you please tell me what values does your testbench has in:
 c_symbols_per_beat
 c_sym_width
 c_data_word_size
And how is the t_slv_array that you are using constrained?

Best regards,
Erick

carmen1403, 2020-03-11:
Hello,

Using 32-bit data bus:
(c_sym_width): 8
(c_data_word_size):32
(c_symbols_per_beat): 4

Using 16-bit data bus:
(c_sym_width): 8
(c_data_word_size):16
(c_symbols_per_beat): 2

I’m sending data from an Ethernet generator so I’m not using the transmit function, I only use the receive function. I don’t need to create a t_slv_array.
I show you one of my 32-bit interfaces.

avalon_st_vvc_if_eth_gen_out : inout t_avalon_st_if(channel(0 downto 0),
                                                            data(31 downto 0),
                                                            data_error(0 downto 0),
                                                            empty(1 downto 0)
                                                            ) := init_avalon_st_if_signals(false, 1, 32, 1, 2)

UVVM, 2020-03-11:
Alright, and how are you setting the parameters data_array_len and data_word_size in the avalon_st_receive() procedure?
I'm just asking because we've tested with 16-bit and 32-bit interfaces before so I'm wondering if there's something else going on.

Best regards,
Erick

carmen1403, 2020-03-11:
For 32-bit:
data_array_len = 728

avalon_st_receive(AVALON_ST_VVCT, 1, 728, 32, "\nFrom LET Framer Test Harness: Receive data in VVC. -> LET_FRAMER TO LET_DEFRAMER");

For 16-bit:
data_array_len = 504

avalon_st_receive(AVALON_ST_VVCT, 3, 504, 16, "\nFrom LET Deframer Test Harness: Receive data in VVC. -> LET_DEFRAMER");

UVVM, 2020-03-11:
I don't see anything wrong with your setup. What's the failing message when you use the original code or how are you checking that it fails?

Best regards,
Erick

carmen1403, 2020-03-11:
Now I’m running my setup with the original function from your GitHub and I get this message:
The AVALON_ST_VVC,1 is the instance for 32-bit and I want data_array_len=728.

# UVVM: =========================================================================================================================================================================
# UVVM: ***  ERROR #1  ***
# UVVM:          1870 ns   AVALON_ST_VVC,1
# UVVM:                    avalon_st_receive(728 sym)=> Failed. End of packet set at symbol #728. '
# UVVM:                    From LET Framer Test Harness: Receive data in VVC. -> LET_FRAMER TO LET_DEFRAMER'  [2]
# UVVM:
# UVVM: Simulator has been paused as requested after 1 ERROR
# UVVM: *** To find the root cause of this alert, step out the HDL calling stack in your simulator. ***
# UVVM: *** For example, step out until you reach the call from the test sequencer. ***
# UVVM: =========================================================================================================================================================================

UVVM, 2020-03-12:
Hi,

It seems like the end of packet is being set too soon. Is your Ethernet generator sending 728 symbols or 728 32-bit words?

Best regards,
Erick

carmen1403, 2020-03-12:
The Ethernet Generator is configured like this:  eth_length=728 and it provides 32-bit data. So it generates 728 symbols (bytes) and in the output you have 4 symbols each time.

UVVM, 2020-03-12:
Alright, then I think you should use:
avalon_st_receive(AVALON_ST_VVCT, 1, 182, 32, "\nFrom LET Framer Test Harness: Receive data in VVC. -> LET_FRAMER TO LET_DEFRAMER");

From the VVC QuickRef:
data_array_len: Length of the data_array expected to be received (number of words).
data_word_size: Size of the data words in the data_array expected to be received.

So if you expect to receive 728 bytes in a 32-bit interface then you should use
data_array_len: 728/4 = 182 (words)
data_word_size: 32 (bits)

carmen1403, 2020-03-13:
Ok, Maybe it was a bad interpretation of mine. Nevertheless, I tried this command with the original function available on GitHub and I get this error.

# UVVM: =========================================================================================================================================================================
# UVVM: ***  ERROR #1  ***
# UVVM:          1860 ns   AVALON_ST_VVC,1
# UVVM:                    avalon_st_receive(182 sym)=> Failed. End of packet not set at last valid transfer. '
# UVVM:                    From Test Harness: Receive data in VVC'  [2]
# UVVM:
# UVVM: Simulator has been paused as requested after 1 ERROR
# UVVM: *** To find the root cause of this alert, step out the HDL calling stack in your simulator. ***
# UVVM: *** For example, step out until you reach the call from the test sequencer. ***
# UVVM: =========================================================================================================================================================================

UVVM, 2020-03-13:
That means the received packet is longer that the 182 words.
I noticed that on the previous error message it says: "End of packet set at symbol #728." which would mean that the Ethernet generator is sending 729-732 symbols (since symbol index starts at 0).

UVVM, 2020-03-19:
Hi,

Did you find out what was the issue or is there anything else we can help you with?

Best Regards,
Erick

carmen1403, 2020-04-27:
Hi Erick,

I update the UVVM-master library in my project and using this line:

`avalon_st_receive(AVALON_ST_VVCT, 1, 182, 32, "\nFrom LET Framer Test Harness: Receive data in VVC. -> LET_FRAMER TO LET_DEFRAMER");`

now it works for me. 

Thank you so much.

UVVM, 2020-04-28:
Hi,

That's great! Thank you for letting us know.

Best Regards,
Erick

closed 2020-04-28 by UVVM