UVVM/UVVM #104: How can C_USE_STD_STOP_ON_ALERT_STOP_LIMIT be overriden without change in source code?

chevillotn, 2020-03-27:
In order to include my testbenches in GitLab CI, i need the output of questasim to return an error code if UVVM test fails (on check_value or any other primitive).
With default setting C_USE_STD_STOP_ON_ALERT_STOP_LIMIT := true, the return value is 0.
If i set it to false then i get the expected != 0 value.

UVVM is an git submodule in our project and i don't wish to change the UVVM source code.
How can this constant be overriden from outside of UVVM? (we use VHDL2008).

Thanks!

UVVM, 2020-04-01:
Hi,
Is there a method for changing the default setting of C_USE_STD_STOP_ON_ALERT_STOP_LIMIT on runtime you need?

Best regards
Marius

chevillotn, 2020-04-01:
Hi Marius,
Sorry i don't get your point. When a UVVM check_value fails for example it will use the primitive selected by C_USE_STD_STOP_ON_ALERT_STOP_LIMIT at compile time. That's how UVVM code is implemented.
That cannot be changed at runtime.

I guess the only solution to this is to have a local copy of this adaptations package and to not compile the UVVM provided one. This is what was implemented in our project where adaptations package file is patched on the fly before compiling.
I wished there was a cleaner VHDL way of doing it.

Regards,
Nicolas.

UVVM, 2020-04-01:
Hi Nicolas,
The adaptations_pkg is as you probably know for users to make local adjustments to UVVM. Unfortunately we can not change the C_USE_STD_STOP_ON_ALERT_STOP_LIMIT to a shared variable as this would break backwards compatibility.
Could you do a GitHub pull and merge with your local version of UVVM, or setup a Python script to make the adjustment in adaptations_pkg to suit your needs?

Best regards,
Marius

chevillotn, 2020-04-01:
Isn't that what i just mentioned above:
"This is what was implemented in our project where adaptations package file is patched on the fly before compiling."
So yes we have this done already.

I'm afraid not breaking backward compatiblity is a little bit light argument. I will use dirty patches then to your code.

Thanks

closed 2020-04-01 by chevillotn