UVVM/UVVM #105: Error injection VVC:

carmen1403, 2020-04-27:
Hi, 

I developed the test bench and test harness needed for test my system using UVVM, for now, I'm using the `clock_generator` and the `avalon_st` VVC (there are instantiated several times). 

Also, I developed a fault injection script for test the system under a stuck-at-0, stuck-at-1 and SEU (Single Event Upset) faults using the TCL commands provided by ModelSim. Thus, I run the simulation from the TCL using this script and at some point it injects the faults into the signal that I choose from the UVVM system. This is working for me but I noticed that UVVM has got an error injection VVC which provides these faults model among other such as invert and delay but I have some doubts and troubles working now with it:

1) This VVC works with the signal events (for example with rising edge) so I am not able to generate this faults and error whenever I want because it's mandatory to wait for, at least, the first event. Then if you set the `interval=2, 3 or 4` you can manage the faults in different clock events. 

2) This VVC needs to be instantiated for each signal so this provokes a lot of overwork, specially when you have a complex system with several signals. This can be managed with the `generate` function but it suppose an overwork and it's difficult to managed it in the test bench if you have, for example, 50 signals that you want to test.

3) This VVC it's fine using it with the signals at high level with the other VVC signals that you have instantiated, for example, with the `avalon_st`. Nevertheless, if you want to test signals inside the architecture, the modules and at a register level this is not so easy. In my case, for get a signal inside the UVVM system I used the `alias` type.

Probably you have ideas regarding this things because you have an overall image of all the UVVM library developed. 

Thank you.

UVVM, 2020-05-06:
Hi,

1. The EI VIP is event driven and will not work for this purpose. But this could be managed by a error injection process in the test harness, which can be controlled with generics and signals from the testbench.

2. For a system with lots of signals there can be some work to setup all the EI VIPs if error injection is wanted on all the signals. Note that the EI VIP has support for SLVs as well.

3. For provoking errors inside the DUT architecture, e.g. to imitate SEUs, there is no other alternative than to alias/reference the signal inside the DUT and use the force and release VHDL commands.

We have demonstrated protocol aware error injection and randomisation in the SBI VVC and UART VVC, and these concepts can be adapted to the Avalon ST VVC as well. Have a look at the UVVM demo testbench located in the bitvis_vip_uart/tb folder for a simple demonstration.

Does this answer your questions?

Best regards,
Marius

carmen1403, 2020-05-06:
Hello Marius,

Yes, this answer my questions. 
Thanks for your time!

UVVM, 2020-05-06:
Answer accepted.

closed 2020-05-06 by UVVM