UVVM/UVVM #106: Code Coverage using xConstrRandFuncCov (OSVVM)

carmen1403, 2020-04-27:
Hi 

UVVM has integrated the `Coverage` and `Random` libraries from OSVVM. I have the verification test bench and test harness for test the system using UVVM and now I would like to check and analysis the code coverage in my system, i.e. what percent of the code is unused and untested. 

I checked the `Coverage` user guide but it seems like it is only available for use when you define you own coverage model and you want to reach it, like the ALU shown as a example in the user guide. So, I would like to know if there is the possibility to check the complete `code coverage` of all the system by using this library? 

Thank you

UVVM, 2020-04-29:
Hi,

There are no libraries available to do code coverage (only functional coverage), as code coverage must be handled by the simulator. The simulator then ticks off for instance every statement that is executed in your code. 
Your simulator may have code coverage included, but it could also be that you need an additional license to use code coverage. 
Applying code coverage is dead simple and requires no additional coding. You only have to set some switches when compiling/simulating. Check out relevant info on this from for instance Mentor or Aldec.

- Espen

carmen1403, 2020-05-06:
Hello Espen,

For code coverage in ModelSim I need a license this is why I was thinking to use this library.
Thanks for the information.

closed 2020-05-06 by UVVM