UVVM/UVVM #108: Continue execution in avalon_st_receive

carmen1403, 2020-05-13:
Hi,

I'm using the Avalon ST VVC for receive information and I'm forcing an error so the execution it's stopped by UVVM:

```
# UVVM: =========================================================================================================================================================================
# UVVM: ***  ERROR #2  ***
# UVVM:         12940 ns   AVALON_ST_VVC,2
# UVVM:                    avalon_st_receive(253 words/506 sym)=> Failed. End of packet set at symbol #502. '
# UVVM:                   Receive data in VVC.'  [27]
# UVVM: 
# UVVM: Simulator has been paused as requested after 2 ERROR
# UVVM: *** To find the root cause of this alert, step out the HDL calling stack in your simulator. ***
# UVVM: *** For example, step out until you reach the call from the test sequencer. ***
# UVVM: =========================================================================================================================================================================
```

Then, I would like to continue the execution after this forced error in order to see the behavior of the system (if it returns to the normal activity or not) but I get this error:

`# Fatal error in ForLoop loop at ../src/libraries/UVVM-master/bitvis_vip_avalon_st/src/avalon_st_bfm_pkg.vhd line 594`

Is possible to delete this command from the cmd_queue and continue the normal execution of the test bench? Run the test bench, force an error which breaks the execution and then continue the execution of the test bench

UVVM, 2020-05-14:
Hi,

You should be able to continue the execution if you increment the alert stop limit, e.g.
`increment_expected_alerts_and_stop_limit(ERROR, 1);`
Or you can set it to never stop at the beginning of your testbench:
`set_alert_stop_limit(ERROR, 0)`

However since you're getting a fatal error I'm not sure there's a way around that. Does it happen immediately after the avalon_st_receive failed?

How are you forcing the error? Could you describe what values of data_width and symbol_width are you using?

Best regards,
Erick

carmen1403, 2020-05-15:
Hi,

Thank you, this is exactly what I was looking for `set_alert_stop_limit(ERROR, 0)`

UVVM, 2020-05-15:
Hi,

That's great. You can also check the uvvm_util/doc/util_quick_ref.pdf for more information on this and other common UVVM functions.

Best regards,
Erick

closed 2020-05-15 by UVVM