UVVM/UVVM #110: axilite_check problem

mergani, 2020-05-23:
Hi everybody,
When I tried to read data from registers on axilite, I see below message.
Both expected and got mesaage same, but it failed.
Do I miss something? I checked waveform interface, it seems no problem.
bitvis_vip_axilite version 2.6.1

Thanks!

```
UVVM: 
UVVM: =========================================================================================================================================================================
UVVM: ***  ERROR #1  ***
UVVM:      723857.5 ns   AXILITE_VVC,1
UVVM:                    axilite_check(A:x"0", x"0000010C")=> Failed. slv Was x"0000010C". Expected x"0000010C".
UVVM:                    'Check data from TMP112_MEAS from axi interface'  [22]
UVVM: =========================================================================================================================================================================
UVVM: 
UVVM:
```

UVVM, 2020-05-25:
Hi,

This is strange. Could you change the following line in the axilite_check() procedure inside axilite_bfm_pkg.vhd (line 638)?

`alert(alert_level, proc_call & "=> Failed. slv Was " & to_string(v_data_value, HEX, AS_IS, INCL_RADIX) & ". Expected " & to_string(data_exp, HEX, AS_IS, INCL_RADIX) & "." & LF & add_msg_delimiter(msg), scope)`;

Replace data_exp for v_normalized_data and show me the ERROR log again please.

Best regards,
Erick

mergani, 2020-05-25:
Hi, 
I changed line 638 to:
`alert(alert_level, proc_call & "=> Failed. slv Was " & to_string(v_data_value, HEX, AS_IS, INCL_RADIX) & ". Expected " & to_string(v_normalized_data, HEX, AS_IS, INCL_RADIX) & "." & LF & add_msg_delimiter(msg), scope);`

But still I got same error message.

UVVM, 2020-05-25:
Ok, try removing the extra parameters in the to_string() functions.

`alert(alert_level, proc_call & "=> Failed. slv Was " & to_string(v_data_value) & ". Expected " & to_string(v_normalized_data) & "." & LF & add_msg_delimiter(msg), scope);`

mergani, 2020-05-25:
Error message changed to :
```
=========================================================================================================================================================================
UVVM: ***  ERROR #1  ***
UVVM:      633857.5 ns   AXILITE_VVC,1
UVVM:                    axilite_check(A:x"0", x"0000010C")=> Failed. slv Was 00000000000000000000000H0000HH00. Expected 00000000000000000000000100001100.
UVVM:                    'Check data from TMP112_MEAS from axi interface_1'  [22]
UVVM: =========================================================================================================================================================================
```

UVVM, 2020-05-25:
This is happening because the axilite_check() procedure compares the received value from your DUT which uses 'H' and 'L' against the x"0000010C" value which uses '1' and '0'.
Currently there's no option in the Axilite BFM to allow the comparison between 'H' and '1' to be true, however you can temporarily fix this by using the **to_01()** function from numeric_std like this:

```
if v_normalized_data(i) = '-' or to_01(v_normalized_data(i)) = to_01(v_data_value(i)) then
  v_check_ok := true;
else
  v_check_ok := false;
  exit;
end if;
```

Best regards,
Erick

mergani, 2020-05-25:
Hi,
I reverted line 638 to original line:
`alert(alert_level, proc_call & "=> Failed. slv Was " & to_string(v_data_value, HEX, AS_IS, INCL_RADIX) & ". Expected " & to_string(data_exp, HEX, AS_IS, INCL_RADIX) & "." & LF & add_msg_delimiter(msg), scope);`
And changed line 629 to:
`if v_normalized_data(i) = '-' or v_normalized_data(i) = to_01(v_data_value(i)) then`
**It worked without error message.** 

To figure out the root cause I would like to mention about my test procedure:
1. DUT reads data on I2C bus
2. DUT writes these data to registers
3. Testbench reads data registers on axilite
Maybe, error is about I2C data to vector convertion issue. 

Thanks for all.

UVVM, 2020-05-26:
Hi,

That's great.
Please see my answer in the UVVM forum: https://forum.uvvm.org/t/axilite-check-problem/63/2?u=erick

Best regards,
Erick

closed 2020-05-26 by UVVM