UVVM/UVVM #111: Cannot use short version of spi_master_transmit, spi_master_receive, spi_master_transmit_and_check

lars555, 2020-06-26:
Hello,
Thanks a lot for UVVM.
After your presentation on FPGA Verification Day 2019 I looked very promising I wanted to use it. 
Now I started to use it and I am sure it will boost my verification. What's nice I integrated it with my previous TB and BFMs and it works :).

I have small remark to SPI BFM.
      
According to SPI BFM Quick Ref there is possible to use spi_master_transmit, spi_master_receive, spi_master_transmit_and_check with shorter version (without passing constant config, scope,msg_id_panel etc. parameters).
In sequencer I call:

        spi2 <= init_spi_if_signals(C_SPI_BFM_CONFIG_PROJ_DEFAULT);
        spi_master_transmit(x"00", "Sending data to Peripheral 2",spi2);

but those functions uses default constant C_SPI_BFM_CONFIG_DEFAULT defined in spi_bfm_pkg with SPi bit time -1 and it causes error 'SPI Bit time was not set in config'.

So, as workaround I pass all parameters to avoid this error.

        spi2 <= init_spi_if_signals(C_SPI_BFM_CONFIG_PROJ_DEFAULT);
        spi_master_transmit(x"00", "Sending data to Peripheral 2",spi2, RELEASE_LINE_AFTER_TRANSFER, C_SCOPE, shared_msg_id_panel, C_SPI_BFM_CONFIG_PROJ_DEFAULT);
 
The same is for spi_master_receive, spi_master_transmit_and_check. 

Shorter version would look more readable.

Best Regards,
PWa

mariuselv, 2020-06-29:
Hi, So happy to hear you find UVVM useful!
This SPI BFM issue you have come across certainly looks like a bug. I will investigate this further and see how this can be solved.

Best regards,
Marius

mariuselv, 2020-06-30:
Hi Lars,
I've have now had a look at the BFM and the answer to your question is somewhat complex. For the BFMs that are setting up a clock signal we have no idea of what your clock period should be, so we have decided to use -1 ns as a default so that we can easily detect and warn you if you forget to set a clock period. In your case, where you want to use the BFM with a minimum of parameters in the BFM call, this will not be possible because the BFM will use the default values, i.e. bit period of -1 ns, as you correctly have stated. As of now there is only two possible solutions:

1. Edit the default BFM config in the BFM package and explicitly set the correct clock period for your design. 
2. Make a local overload which takes the desired number of parameters, and which again calls the SPI BFM call with the remaining parameters. This will of course require that you define your own BFM configuration constant of type t_spi_bfm_config with the correct clock period, and use this configuration in the BFM call.

You define a BFM configuration in your testbench like this:

`  constant C_SPI_BFM_CONFIG : t_spi_bfm_config := (
    CPOL                    => '0',
    CPHA                    => '1',
    spi_bit_time          => 200 ns,
    ss_n_to_sclk         => 301 ns,
    sclk_to_ss_n         => 301 ns,
    inter_word_delay  =>   0 ns,
    match_strictness => MATCH_EXACT,
    id_for_bfm           => ID_BFM,
    id_for_bfm_wait   => ID_BFM_WAIT,
    id_for_bfm_poll   => ID_BFM_POLL
    );`

I will update the quick reference to clarify this issue.

Br,
Marius

lars555, 2020-07-09:
Marius,

Thank you for your investigation and quick response.

I've used bitvis_vip_spi. Besides useful instantiation feature it let using SPI project specific config without need of changing BFM or creating local overload (it uses vvc_config.bfm_config which is set to project defaults C_SPI_BFM_CONFIG_PROJ_DEFAULT in instantiation). So I use shorter version of spi_master_... procedures calls in my code now. 

Best regards

Pawel 

_alias Lars555_

lukipedio, 2020-07-25:
@mariuselv is it a bug or a feature? It seems like it is a precise UVVM design choice.
Are you thinking about an alternative method?

mariuselv, 2020-07-25:
Hi Luca,
it is provided as an alternative for calling the BFM, but it requires that modifications are made to the BFM local default configuration by specifying spi_bit_time (and other settings if applicable). The upcoming release will have more information regarding this as we see there might be misunderstandings with how this is intended to be used.
We are looking into other methods and are discussing how this could best be implemented. We want a solution that we can use in all our BFMs and that will not be a showstopper later on.

mariuselv, 2020-08-31:
August release have information regarding local BFM overloads

closed 2020-08-31 by mariuselv