UVVM/UVVM #112: Re-factor documentation

GlenNicholls, 2020-07-14:
I have used UVVM and found that the documentation is very solid. However, the PDF's aren't very accessible for quick navigation as it requires me to open multiple PDFs that I usually always keep open while I am trying to find more specific information. Because of this, I have resorted to referencing the code instead. I am proposing that the documentation is changed such that instead of doing partial documentation in README's and PDFs, everything is moved to readthedocs instead, similar to [VUnit](https://vunit.github.io/index.html). 

In my opinion, I shouldn't need to navigate the code to understand how to use UVVM, and VUnit for example has done a great job of putting the documentation on the web. This facilitates much faster navigation, and after using it for a while, it is obvious where the desired information is located. Instead of navigating many PDFs trying to find something specific, I can either search Google, or I can click a few ToC links to get the information immediately. 

Again, y'all have done a fantastic job with documentation, I just feel that moving it to readthedocs (like most FOSS projects) will allow wider adoption and less effort when trying to find specific information.

GlenNicholls, 2020-07-15:
Also, the other benefit of getting the PDFs out of the repo is that it will reduce the repo clone time since they're fairly large files.

EspenTallaksen, 2020-07-18:
Good suggestion and points. Only problem is that it is quite a time consuming task. We did actually check this out on one case only a few weeks ago and the automatic conversion was quite good, but terrible when it came to converting the tables...
I guess we can re-evaluate this when we have more time available. At the moment we are "unfortunately" quite busy. Feel free to suggest approaches to speed this up.

- Espen

GlenNicholls, 2020-07-20:
What did you use to convert the PDFs to that format? Did you use your current PDFs directly when trying that?

mariuselv, 2020-08-07:
Hi,
If I remember correctly we used Pandoc with PDF to rst direct conversion. There might be some switches that could improve the result, we only performed a quick test without exploring the possibilities of the tool.

UVVM, 2023-03-27:
In transition now to html.

closed 2023-03-27 by UVVM