UVVM/UVVM #113: Configure CI tasks for UVVM

GlenNicholls, 2020-07-15:
CI is really nice for regular and continuous testing of code and regressions. I'm not sure if BitVis has internal tests for UVVM, but they should really be public such that integration with other tools is easier. This would hopefully lead to standardization across VHDL projects. Things like source management, tooling bugs, etc., vary wildly between projects and it makes it difficult to create tools that utilize many open source projects as it requires either a user or separate project to deal with these inconsistencies. I am proposing that UVVM set up GitHub actions to regularly run UVVM tests against the GHDL simulator and preferably provide a bash script to compile the code with GHDL.

References for creating compilation scripts for GHDL can be found in https://github.com/ghdl/ghdl/tree/master/libraries/vendors

EspenTallaksen, 2020-07-18:
Yes, we do have internal testbenches, and currently only a subset has been released as example or demo testbenches. We are however working on releasing our internal testbenches, and we will most probably do this already in August. July is pretty much vacation time in Norway...

- Espen

mariuselv, 2020-08-31:
Maintenance testbenches are available in UVVM_SUPPLEMENTARY repository

closed 2020-08-31 by mariuselv