UVVM/UVVM #114: Use of I2C BFM hints

lukipedio, 2020-07-25:
I am quite new to UVVM, so sorry for the silly question maybe.

I have a running I2C master, which works quite well in hardware.
Now I have to add some clock stretching feature or recovery mode, so I decided to implement a testbed based on SPI BFM,
so that I can also practice with the framework.

So my testbench is composed by the I2C master, an i2C slave model and a call to i2c_slave_receive() to sniff what is passing on the bus.
I assume i2c_slave_receive() cannot be used as a slave model, right?

I am generating an access to a slave mapped to I2C address 0x44. 

The BFM is issuing an error stating:
[image]

The simulation waveform is:
[image]

Any fault on my side?

lukipedio, 2020-07-27:
I realize the information I give you are very limited.

I just call:
```
        i2c_if <= init_i2c_if_signals(VOID);
        i2c_slave_receive(v_data_out, "Receive from Master");
```

with local defined config structure:
```
  constant C_I2C_BFM_CONFIG : t_i2c_bfm_config := (
    enable_10_bits_addressing       => false,
    master_sda_to_scl               => 20 ns,
    master_scl_to_sda               => 20 ns,
    master_stop_condition_hold_time => 20 ns,
    max_wait_scl_change             => 10 ms,
    max_wait_scl_change_severity    => failure,
    max_wait_sda_change             => 10 ms,
    max_wait_sda_change_severity    => failure,
    i2c_bit_time                    => cc_halfp*2*clk_period,
    i2c_bit_time_severity           => failure,
    acknowledge_severity            => failure,
    slave_mode_address              => "0001000100",
    slave_mode_address_severity     => failure,
    slave_rw_bit_severity           => failure,
    reserved_address_severity       => warning,
    match_strictness                => MATCH_EXACT,
    id_for_bfm                      => ID_BFM,
    id_for_bfm_wait                 => ID_BFM_WAIT,
    id_for_bfm_poll                 => ID_BFM_POLL
    );
```
i2c_bit_time is 20*2*5 = 200 ns which is my SCL bit time.

mariuselv, 2020-07-27:
Hi Luca,
You are right that the BFM can not be used as a model. 
I guess you are using a local overload in your testbench - can you paste more code from your testbench? 

Br,
Marius

lukipedio, 2020-07-27:
Hi Marius,

  here's the complete testbench

[i2c_master_tb.txt](https://github.com/UVVM/UVVM/files/4981473/i2c_master_tb.txt)

mariuselv, 2020-07-27:
Can you set the BFM config entry "enable_10_bits_addressing <= True" and try again?

lukipedio, 2020-07-27:
[image]

mariuselv, 2020-07-29:
Hm, I'm trying to figure out what could be the root of this error. I guess you are using 7 bit addressing then, so you have to set the enable_10_bits_addressing back to false. From your initial post it seems that the BFM is missing the upper bit of the address. Which version of UVVM are you using? I do not think that there has been any major changes to the I2C VIP in a while so unless you are using a very old version the source code should be mostly identical.

lukipedio, 2020-07-29:
I am using UVVM tagged v2020.06.23
[image]

Is there any change that I must call i2c_slave_receive() at a precise time?
I am issuing the call just after init_i2c_if_signals(VOID), as you can see from the testbench.
At the beginning of the test there is a reset pulse, so init_i2c_if_signals(VOID) is called  just 5 clock cyclose after the reset returns high.

mariuselv, 2020-07-29:
No, the BFM waits for a SDA to be set low, then SCL, before sampling the address and the R/W bit. There is a timeout which will generate an alert if there is no interface activity when the BFM is waiting, in your case within 10 ms of SDA and additional 10 ms for the SCL line, so I do not think that is the problem. Is it a write access you are requesting?

mariuselv, 2020-07-29:
I think the problem is that your SDA line is 'U' at the beginning, while the BFM is waiting for it to be '1', i.e. free, so the data sampling actually do not start until after your first '1' on the SDA line. So if you have a look at the wave diagram below the red circle indicates when the BFM thinks the I2C line is free, and starts sampling the data byte from the next CLK high period.
[image]

So the BFM does:

1. Wait for SDA to be idle / '1'
2. Wait for SCL to be idle / '1'
3. Wait for SDA to be active / '0'
4. Wait for SCL to be active / '0'
5. Start sampling byte

lukipedio, 2020-07-29:
[image]

Now it works! That was the trick!

The 'U' at the beginning are due to the iobuf/obuft instances inside the master (there shouldn't be there!!).

I have introduces some spare SCL cycles at the beginning, so the start condition is found later on.
I have add this small simulation setup on github (https://github.com/lukipedio/i2c_master_uvvm) .

Thanks for helping!

mariuselv, 2020-07-29:
Thats great! 
Then I will close this issue. If you have more questions please do not hesitate to post them on our forum at https://forum.uvvm.org

Kind regards,
Marius

closed 2020-07-29 by mariuselv