UVVM/UVVM #115: stable GPIO value for dedicated time

lars555, 2020-09-02:
Hello,

During verification there is often need to check that GPIO signal keeps expected value for proper time. It would be very useful to have it. I don't see possibility to use GPIO VIP API for this need now.

Please consider extension of VVC gpio_check procedure with additional stable_time parameter set to 0 ns by default. 
If gpio_vvc_if does not keep value provided by data_exp during stable_time then raise an alert.

Best Regards

Pawel

UVVM, 2020-09-10:
Hi Pawel,

Thank you for the suggestion, I will create a ticket so that we can implement this for the next release.

Best Regards,
Erick

mariuselv, 2020-10-06:
Fixed in release v2_v2020.10.06

closed 2020-10-06 by mariuselv