UVVM/UVVM #116: Test status is not present in ModelSim coverage HTML report

lars555, 2020-09-15:
Hello,

Test status is not present in merged ModelSim coverage HTML report. This summary report is very useful for execution of many test configurations (many test sequencers for the same DUT).
Coverage test status uses severity_level of report statement to indicate test status (pass, error, warning, failure).
Procedure alert() from uvvm_utils does'nt call report.
To have clear indication of test status in summary report please add it. 

Possible change of alert procedure implementation:

```
   ... 
   if alert_level /= NO_ALERT then
      pot_initialise_util(VOID);        -- Only executed the first time called

      case alert_level is 
        when error =>
         report "Error. See UVVM alert message ." severity error;
        when warning => 
         report "Warning. See UVVM alert message." severity warning;
        when failure =>
           report "Failure. See UVVM alert message." severity failure;
        when others =>
      end case;
   ...
   end if;
```

Of course it could be configurable e.g. in adaptation_pkg to enable/disable alert report calls.

Best regards,

Pawel

mariuselv, 2020-09-21:
Hi Pawel.
Thanks for the feedback, I will add a ticket so we can make a fix.

Best regards,
Marius

fisketavlingnu, 2022-10-26:
Simulator test status takes all sources of errors into account (such as assertions, $error,$warning, vhdl severity).  As it is now you can actually get that the UVVM tb say that the simulation passed even if a library component signalled error.

So either this:
[image]
or this:
[image]

If you are running automated regression this teststatus is the one used for checking if a simulation passed or not.  Because you need to check that there are no failing assertions etc.

Here is also a simple tcl procedure to show the teststatus (in modelsim/questa) that can be run after the simulation is finished:

```
proc show_teststatus {} {     
    set teststatus [string trim [lindex [split [coverage attribute -name TESTSTATUS] =] 2]] 
    if {$teststatus == 0} {
        echo "# ** Note:#########################################################################"
        echo "# ** Note:Simulator indicates that simulation PASSED (TESTSTATUS=$teststatus)"
        echo "# ** Note:#########################################################################"
    } elseif {$teststatus == 1} {
        echo "# ** Note:#########################################################################"
        echo "# ** Note:Simulator indicates that simulation PASSED WITH WARNINGS (TESTSTATUS=$teststatus)"
        echo "# **       TSTAT_REASON=[lindex [ lindex [coverage attribute -name TSTAT_REASON -concise -tcl] 0] 0]"     
        echo "# ** Note:#########################################################################"
    } elseif {$teststatus > 1} {
        echo "# ** Error:#########################################################################"
        echo "# ** Error:Simulator indicates that simulation FAILED  TESTSTATUS > 1 ACTUAL=$teststatus"
        echo "# **       TSTAT_REASON=[lindex [ lindex [coverage attribute -name TSTAT_REASON -concise -tcl] 0] 0]"     
        echo "# ** Error:#########################################################################"
    }  else {
        echo "# ** Error:#########################################################################"
        echo "# ** Error:Simulator indicates unknown teststatus    TESTSTATUS = $teststatus"
       echo "# ** Error:#########################################################################"
    }
}
```

EspenTallaksen, 2022-10-28:
Hi,
This has been tested a lot, so I assume the problem could be a delta cycle racing issue. (Between the VVC and the sequencer)
Try to wait 1 ns to see if that works. 

-- Espen

fisketavlingnu, 2022-10-31:
Providing a testcase through email that illustrates the problem. Notice that I check teststatus twice with 1 ns in between. 
To run the testcase, unpack. go to uvvm_teststatus/bitvis_uart/sim/
Update the UVVM_HOME variable in the makefile and type "make batch".

The suggested change in the beginning of this issue will solve the problem.
Also notice how fast this compile and recompiles. 
Checkout qrun in the manual for Questasim for more information.

fisketavlingnu, 2022-10-31:
Providing a testcase that illustrates the problem. Notice that I check teststatus twice with 1 ns in between.
To run the testcase, unpack. go to uvvm_teststatus/bitvis_uart/sim/
Update the UVVM_HOME variable in the makefile and type "make batch".

The suggested change in the beginning of this issue will solve the problem.
Also notice how fast this compile and recompiles.
Checkout qrun in the manual for Questasim for more information.

 

 

Testcase attached

 

BR

Mikael

fisketavlingnu, 2023-03-27:
Providing a testcase that illustrates the problem. Notice that I check teststatus twice with 1 ns in between.
To run the testcase, unpack. go to uvvm_teststatus/bitvis_uart/sim/
Update the UVVM_HOME variable in the makefile and type "make batch".
The suggested change in the beginning of this issue will solve the problem.
Also notice how fast this compile and recompiles.
Checkout qrun in the manual for Questasim for more information.

Testcase attached
[uvvm_teststatus.tar.gz](https://github.com/UVVM/UVVM/files/11077843/uvvm_teststatus.tar.gz)

BR
Mikael

UVVM, 2026-10-16:
This is now implemented and will be part of the next release.

UVVM, 2026-10-27:
Released in v2 2026.10.27

closed 2026-10-27 by UVVM