UVVM/UVVM #117: AXI_LITE BFM configuration record changed

stfarley, 2020-10-22:
With 2020.10.06 the AXILITE BFM configuration has changed. The protection_setting (now t_axprot) fields have changed.
The axilite_bfm_Quickref has not been updated to reflect this.

BTW changes like this should be considered carefully. I just went through my entire development system updating and checking the last set of changes to this. Now it has to be done again.

mariuselv, 2020-10-23:
Hi,
I had to check with the developer what these changes were and I was told they were done because of naming typos and to correct values that were not according to the standard.
Unfortunately these changes was not mentioned in the CHANGES.TXT or updated in the quick reference, nor catched in release process. I have made a ticket to ensure that the quick reference is updated with the next release.

We always try to make the changes backwards compatible, but unfortunately this one slipped through. 

Best regards,
Marius

stfarley, 2020-10-23:
Hi,

Thanks for the response and for all of the hard work that everybody has been doing on UVVM. 

Steve

closed 2020-12-08 by mariuselv