UVVM/UVVM #118: vip_axilite fails in newer modelsim versions

Obliged, 2020-10-26:
The v2 2020.10.06 release which adds axilite_channel_handler_pkg fails with a segmentation error i in newer versions of Modelsim DE. 

This is the error message
```
# Attempting stack trace sig 11
# Signal caught: signo [11]
# vsim_stacktrace.vstf written
# Current time Fri Oct 23 15:20:42 2020
# ModelSim DE Stack Trace
# Program = vsim
# Id = "2020.4"
# Version = "2020.10"
# Date = "Oct 13 2020"
# Platform = "linuxpe"
# Signature = a791d2643986943217c290b2b32443f4
# 0    0x08304195: '<unknown (@0x8304195)>'
# 1    0x0830465f: '<unknown (@0x830465f)>'
# 2    0x08305a99: '<unknown (@0x8305a99)>'
# 3    0xf6ac81e4: '/disk1/jenkins/workspace/kspt_tb_axiLiteRegisters_master/UVVM/bitvis_vip_axilite/src/axilite_vvc.vhd:461'
# 4    0x08379139: '<unknown (@0x8379139)>'
# 5    0x085a416a: '<unknown (@0x85a416a)>'
# 6    0xf6ac0193: '/disk1/jenkins/workspace/kspt_tb_axiLiteRegisters_master/kspt/src/support/tb_axiLiteRegisters/Vhdl/tb_axiLiteRegisters.testBench.vhd:87'
# 7    0x08379139: '<unknown (@0x8379139)>'
# 8    0x085a416a: '<unknown (@0x85a416a)>'
# 9    0x08652c05: '<unknown (@0x8652c05)>'
# 10   0x0863ca81: '<unknown (@0x863ca81)>'
# 11   0x0863dbe1: '<unknown (@0x863dbe1)>'
# 12   0x08b293bc: '<unknown (@0x8b293bc)>'
# 13   0x08b1eec9: '<unknown (@0x8b1eec9)>'
# 14   0x092d34eb: '<unknown (@0x92d34eb)>'
# 15   0x092d2ada: '<unknown (@0x92d2ada)>'
# 16   0x092d8063: '<unknown (@0x92d8063)>'
# 17   0x092d817a: '<unknown (@0x92d817a)>'
# 18   0x08816bc9: '<unknown (@0x8816bc9)>'
# 19   0x086a6519: '<unknown (@0x86a6519)>'
# End of Stack Trace

/.. Lines deleted for brevity ../

#    Time: 0 ps  Iteration: 0  Process: /tb_axiliteregisters/i_axilite_vvc/read_address_channel_executor File: /pathtoworkspace/UVVM/bitvis_vip_axilite/src/axilite_vvc.vhd
```

Tested modelsim DE versions:
2020.4 (fails)
2020.1 (fails)
2019.3 (works)

When we checkout v2020.08.21 the testbench works in all the versions listed.
We didn't change anything in the testbench except update our filelist to include axilite_channel_handler_pkg.

UVVM, 2020-10-26:
Hi,

Yes, this seems to be a bug in the simulator related to the unconstrained arrays used in the Axilite VVC when calling the procedures from the axilite_channel_handler_pkg. We will contact Mentor about this.

In the meanwhile, what you can do is to map the signals from the record interfaces directly in each procedure call in the VVC, e.g.
```
write_response_channel_check(msg                => format_msg(v_cmd),
                              clk                => clk,
                              bready             => axilite_vvc_master_if.write_response_channel.bready,
                              bresp              => axilite_vvc_master_if.write_response_channel.bresp,
                              bvalid             => axilite_vvc_master_if.write_response_channel.bvalid,
                              alert_level        => error,
                              scope              => C_CHANNEL_SCOPE,
                              msg_id_panel       => v_msg_id_panel,
                              config             => vvc_config.bfm_config);
```
Which means you have to update the procedures in the axilite_channel_handler_pkg.

You can find the updated files I used for testing this fix in the zip file below to save you some time.
[axilite_modelsim_2020_fix.zip](https://github.com/UVVM/UVVM/files/5438563/axilite_modelsim_2020_fix.zip)

Best regards,
Erick

Ju-fpga-40, 2021-04-02:
I found the same issue with axi_channel_handler_pkg on modelsim2021.1 .
Workaround is  the same, you can find bellow the files that fix the issue.
[axi_src_corrected.zip](https://github.com/UVVM/UVVM/files/6251078/axi_src_corrected.zip)

JulienH

mariuselv, 2021-04-09:
Fixed in v2021.04.08

closed 2021-04-09 by mariuselv