UVVM/UVVM #119: axistream_expect times out immediately (questasim 2019.1)

fransschreuder, 2020-11-12:
Hello,
I am trying to make a testbench for an entity that generates AXI stream packages (32b + 4b tuser).
The problem I am struggling with is that axistream_expect times out immediately with the following message `axistream_expect(8B) while executing axistream_receive=> Failed. Timeout while waiting for valid data. 'expecting axi stream data'  [1]`
This message happens while absolutely nothing happened yet on axistream_vvc_if. (tready is '0', clock is running).

Some relevant code snippets:

`signal axistream_vvc_if : t_axistream_if(tdata(31 downto 0), tkeep(3 downto 0), tuser(3 downto 0), tstrb(3 downto 0), tid(0 downto 0), tdest(0 downto 0)) := init_axistream_if_signals(false, 32, 4, 1, 1);`

```
checker : process
begin
    checker_done <= false;
    await_uvvm_initialization(VOID);
    axistream_expect(AXISTREAM_VVCT, 0, TESTDATA0,  "expecting axi stream data");
    await_completion(AXISTREAM_VVCT, 0, 1200 * C_CLK40_PERIOD);
    checker_done <= true;
    wait;
end process;
```
The VVC
```
            axi_stream_vvc_I: entity bitvis_vip_axistream.axistream_vvc
                generic map (
                    GC_VVC_IS_MASTER => false,
                    GC_DATA_WIDTH => 32,
                    GC_USER_WIDTH => 4,
                    GC_ID_WIDTH => 1,
                    GC_DEST_WIDTH => 1,
                    GC_INSTANCE_IDX => 0,
                    GC_AXISTREAM_BFM_CONFIG => GC_AXISTREAM_BFM_CONFIG
                )
                port map (
                    clk => m_axis_aclk,
                    axistream_vvc_if => axistream_vvc_if
                );
```

fransschreuder, 2020-11-12:
Here is a screenshot of the axistream VVC waveform. It times out on the right of the waveform, while nothing on the axi stream interface has happened yet.
[image]

UVVM, 2020-11-12:
Hi,

From what I can see in the waveform it looks like your checker process executes axistream_expect() (tready = '1') but there is no data coming in (tvalid = '0') so it times out.
The default timeout in the Axistream BFM config is 100 cycles (max_wait_cycles), you can try increasing it accordingly to your test.

Best regards,
Erick

fransschreuder, 2020-11-12:
Hello Eric,
tready was '1' because I set that as the default value in the BFM config. Increasing max_wait_cycles solved the issue.
Thanks!

closed 2020-11-12 by fransschreuder