UVVM/UVVM #120: to_string(t_slv_array) only works for byte arrays

Obliged, 2020-11-12:
I'm using the new axi_vvc with a 256bit wide data channel and ran into an issue with this error message:

```
# ** Fatal: (vsim-3471) Slice range (1 to 783) does not belong to the prefix index range (1 to 84).
#    Time: 22 ns  Iteration: 17  Process: /tb_dmafifoconverter/i_axi_vvc_master/write_data_channel_executor File: /home/hos/pro/work5/UVVM/uvvm_util/src/string_methods_pkg.vhd
# Fatal error in Subprogram to_string at /home/hos/pro/work5/UVVM/uvvm_util/src/string_methods_pkg.vhd line 1278
# 
# HDL call sequence:
# Stopped at /home/hos/pro/work5/UVVM/uvvm_util/src/string_methods_pkg.vhd 1278 Subprogram to_string - -
# called from  /home/hos/pro/work5/UVVM/bitvis_vip_axi/src/axi_channel_handler_pkg.vhd 255 Subprogram write_data_channel_write - -
# called from  /home/hos/pro/work5/UVVM/bitvis_vip_axi/src/axi_vvc.vhd 786 Process write_data_channel_executor - -
```

I've traced the issue down to this line:
https://github.com/UVVM/UVVM/blob/160ce4f28766eb4b7f711c7e0c4948eac4d2e447/uvvm_util/src/string_methods_pkg.vhd#L1252

to_string for a std_logic_vector cannot return max `26` symbols. It can return `10 + 2 * v_val(0)'length` symbols. `v_val(0)'length` might be a problem if this function is supposed to support empty t_slv_arrays.

jorgenkrohn, 2020-11-16:
Hi,

Thank you. We have made a fix to string_methods_pkg.vhd which will be included in the next release.
In the meantime you can use the updated code that I have attached.

[string_methods_pkg.zip](https://github.com/UVVM/UVVM/files/5546245/string_methods_pkg.zip)

Best regards,
Jørgen

mariuselv, 2020-12-08:
Fixed in v2_v2020.12.08

closed 2020-12-08 by mariuselv