UVVM/UVVM #121: vip_axistream : Limited control of random ready low

Obliged, 2020-11-13:
In axistream_bfm_pkg the amount of control you can exert on the configurations `ready_low_duration = C_RANDOM` and `ready_low_at_word_num = C_MULTIPLE_RANDOM` is a bit too low in my opinion.

The random duration from 1 to 5 clock cycles is too low for cases where the master is slow.
https://github.com/UVVM/UVVM/blob/160ce4f28766eb4b7f711c7e0c4948eac4d2e447/bitvis_vip_axistream/src/axistream_bfm_pkg.vhd#L1362

It would be nice to able to set a rate of ready lows. 
Also range of 2 to 6 words between each ready low misses the corner case of ready low for consecutive words.
https://github.com/UVVM/UVVM/blob/160ce4f28766eb4b7f711c7e0c4948eac4d2e447/bitvis_vip_axistream/src/axistream_bfm_pkg.vhd#L1399

--

For the first I suggest adding a new config `ready_low_duration_max` to be used like this:
```
v_ready_low_duration := random(1,config.ready_low_duration_max);
```

For the second, add a new config `multiple_random_rate` and determine the rate of lows like this:
```
  -- Deassert tready multiple random times per transmission
  elsif config.ready_low_at_word_num = C_MULTIPLE_RANDOM
         and random(1, 100) <= config.multiple_random_rate -- set ready low for a percentage
         and v_byte_cnt mod c_num_bytes_per_word  -- check once per word
  then
```

I see from the comment that `v_next_deassert_byte` is explicitly set to not be the next word, but I don't understand why as this seems like a relevant corner case to me. Is there a technical reason?

UVVM, 2020-11-19:
Thank you for your suggestion.
We will implement this in our next release.

Best regards,
Erick

mariuselv, 2020-12-08:
Fixed in v2_v2020.12.08

closed 2020-12-08 by mariuselv