UVVM/UVVM #128: Modelsim 2019.2 and UVVM Simple

bdecelles, 2021-01-13:
My group seems to be moving to Modelsim 2019.2 from Modelsim DE 10.6 due to moving to Vivado 2019.2.  I see the note on UVVM about issues with Modelsim 2019.2.  Can someone tell me what the issues are and if I will face any Modelsim crashes if I am using just the UVVM BFMs and UVVM uvvm_util compiled Lib?  I can move this question to the GitHub forum or another.

mariuselv, 2021-01-14:
Hi, if I remember correctly there was an issue with Modelsim 2019.2 and some of the VVCs. If you run in to some problems you could try with UVVM Light, a published version of UVVM without VVCs.

Br,
Marius

closed 2021-01-14 by mariuselv

bdecelles, 2021-01-14:
Hi 
Thanks for the reply. We might be OK.  We are essentially using UVVM light and taking the BFMs and creating our own Generators. The gens are real simple but work well for our needs. We have about 15 additional interface protocols built up with gen and bfm mostly proprietary. Thanks for the roadmap and details.