UVVM/UVVM #130: Missing linebreak in compile_order files

Paebbels, 2021-01-17:
When reading `compile_order.txt` files, I found that most files miss a final linebreak.

Code to reproduce:
```Bash
while IFS= read -r File; do
  echo "${File%?}"
done < <(cat "bitvis_vip_gmii/script/compile_order.txt")
```

You'll see `../src/vvc_context.vhd` is missing.

mariuselv, 2021-01-17:
Hi,
Thank you for the report. Have you already made the fix and can create a pull-request?

Br,
Marius

Paebbels, 2021-01-17:
@mariuselv I updated the UVVM pre-compile scripts in GHDL, where I found this. I didn't fix anything on UVVM side, so I can't share a PR.

mariuselv, 2021-01-21:
> @mariuselv I updated the UVVM pre-compile scripts in GHDL, where I found this. I didn't fix anything on UVVM side, so I can't share a PR.

OK, I will make the fix.

Paebbels, 2021-04-03:
When is  this fix available?
GHDL got an implementation of `'element` so UVVM can be (fully?) compiled with GHDL.

mariuselv, 2021-04-04:
Hi Patrick,
Hopefully I will manage to get to publish the release in the next week.

mariuselv, 2021-04-09:
Hi,
Please try with the latest release.

Br,
Marius

Paebbels, 2021-04-09:
@mariuselv This looks good. See this CI run: https://github.com/ghdl/extended-tests/runs/2310131430?check_suite_focus=true

Does UVVM have public testcases or examples to run?

mariuselv, 2021-04-12:
@Paebbels great!
There are mostly just some demo TBs for a small selection of the VIPs in the UVVM repository. However there are lots of more tests located in the [UVVM_SUPPLEMENTARY](https://github.com/UVVM/UVVM_SUPPLEMENTARY) repository, which is identical to the UVVM repository, only with all documentation sources, testbenches etc. Unfortunately not all tests in UVVM_SUPPLEMENTARY can be run as they use verification components that we are not legalized to publish due to license restrictions. 

I think GHDL is a great project and that You are doing an impressive work. Please let ut know if you need any assistance with UVVM tests or something else.

mariuselv, 2021-04-28:
This issue has been solved in v2021.04.08

closed 2021-04-28 by mariuselv