UVVM/UVVM #131: Moving bitvis_vip_wishbone to the UVVM repository.

m-kru, 2021-01-21:
Let's face the truth. The idea of [UVVM_Community_VIPs](https://github.com/UVVM/UVVM_Community_VIPs) has not catch on.
The fact, that `bitvis_vip_wishbone` is put there only generates the mess and increases the cognitive complexity.

The `bitvis_vip_wishbone` is in separate repository, however it needs to be copied to the UVVM repository to be compiled.
Because it is not in the UVVM repository by default GHDL `compile-uvvm.sh` script does not support it, so it needs manual modifications. All this is pure waste of time, putting `bitvis_vip_wishbone` into separate repository solves 0 problems, and creates at least 2.

Can `bitvis_vip_wishbone` be moved into the UVVM repository?

One more question.
What is the point of [UVVM_SUPPLEMENTARY](https://github.com/UVVM/UVVM_SUPPLEMENTARY) repository?

Paebbels, 2021-01-21:
UVVM changed the license (hopefully with a license clearance) to Apache License 2.0 so it can be published under the umbrella of IEEE.

As now UVVM is Apache licensed: Why are user contributions are split apart, while the license itself plus the contribution agreements handle how to integrated more code from users into UVVM?

mariuselv, 2021-01-26:
The Wishbone is in beta and require additional work from our side before it can be added to the UVVM repository.
User contributions are kept aside from UVVM in a separate repository as including them in UVVM would require that we review and make tests to verify them, and that the contributors will have to make documentation similar to those shipped with UVVM. We belive this could make contributors not willing to do the additional work and not share their work.

m-kru, 2021-01-26:
@mariuselv 
```
commit 6eb24b30b65f1dc8dd6f0a08c955612e923644e8
Author: UVVM <info@bitvis.no>
Date:   Tue Aug 6 13:01:35 2019 +0200

    added Bitvis VIP Wishbone BETA
```
Beta since 6th August 2019? 

> User contributions are kept aside from UVVM in a separate repository.

Author: UVVM <info@bitvis.no>

mariuselv, 2021-01-26:
I will up the priority in the jira ticket so that someone hopefully can start the work sometime soon.

m-kru, 2021-03-29:
@mariuselv any status update?

mariuselv, 2021-04-04:
Hi,
unfortunately we have not had the time to start on this task yet - our team is small and also working on other UVVM tasks and other projects. Unfortunately, I do not know when we will get started with this task.

Br,
Marius

m-kru, 2021-04-27:
@mariuselv moving it to the UVVM is probably not a lot of work. I can do it, if you will merge my PR.

m-kru, 2021-04-28:
https://github.com/UVVM/UVVM/pull/140

mariuselv, 2021-05-03:
Hi @m-kru,
I think we can have the Wishbone VIP in the UVVM repository, but users have to be notified that it is beta, i.e. it is not fully tested and with missing documentation. This can be done by giving the folder name a "_beta" suffix - bitvis_vip_wishbone_beta and removing all files in the /doc folder.
Any thoughts?

m-kru, 2021-05-03:
Hi @mariuselv
I was thinking the same, but it is already mentioned in a few places that this is beta. In the `versions.txt` one can see:
```
  bitvis_vip_wishbone, version v.0.1.0 BETA
```
There is the BETA suffix. The first number in the version is 0. In semantic versioning it means that this is BETA and potentially everything is unstable and can change.
In the `bitvis_vip_wishbone/VERSION.TXT` one can also see:
```
0.1.0 BETA
```
Having `BETA` in the directory name may imply some adjustment in other projects utilizing UVVM when Wishbone will no longer be BETA. I do not think this BETA suffix is needed.

There is not `doc/` directory in the `bitvis_vip_wishbone`.

m-kru, 2021-05-18:
Moved to UVVM.

closed 2021-05-18 by m-kru