UVVM/UVVM #132: Await_completion for a multi-executor VVC.

roynil, 2021-03-05:
I have an interface:

```
------------        ---------
|  Master  |--req-->| Slave |
|          |<—resp—-|       |
------------        ---------
```

The slave will answer with the response 1 cycle after a request.
I would like to have the possibility to make multiple requests, i.e. the master can request continuously. Likewise if I run my VVC as a slave I should be able to make multiple responses.

For that I have implemented a multi-executor vvc. With 2 executor handlings request & response.

```
—>interp—>cmd_queue —> executor <———> request
                          |
                          —>cmd_queue—>executor<———>response 

```
I would like to have await_completion to be true both for when a request has completed and when a response has completed. I haven’t figured out how or if that would be possible. 

So how should inactive/active be handled? I see that the vvc_generator.py generates a vvc were both executors independently sets inactive/active. But in your vip:s (avalon_mm & axi) you have some gated criteria with however the cmd_queues are empty or not. The criterias are not included for the await_any_completion. Wouldn’t that end up in a different behaviour between await_completion and await_any_completion?

mariuselv, 2021-03-08:
Hi Ronny,
For us to better help you, could you answer this 3 questions?
1. Which VVC are you using? A version you have developed yourself or any of ours?
2. Does the VVC use the VVC activity register, i.e. does it use the old or new await_X() ?
3. Can you clearify why you need an await completion on a request and not just issue multiple requests at the same time?

Br,
Marius

roynil, 2021-03-08:
Hi Marius,
1. Self developed using vvc_generator.py from UVVM 2.11.1
2. It uses the new_await.
3. When acting as a master the response might be something that a sequencer might have to react on. 

My main question is really (perhaps a bit theoretical): what is an await_completion for a multi-executioner VVC? It's clearly not the same for await_any_completion and await_completion right now.

Regards, Ronny

mariuselv, 2021-03-11:
Hi,
You should not be using the await_any_completion() method in with that version of UVVM. The await_any_completion() method was deprecated in the v2 2020.05.19 release of UVVM and you should now use the await_completion() method as described in Common VVC Methods pdf.
Sorry, but I have not had the time to check why you get different behaviors with the two methods, but hope this can help you a bit.

Br,
Marius

UVVM, 2023-03-27:
Hi,
The await_completion() (with no extra options) should always wait until the VVC is inactive - i.e. all internal executors completed.
Does this answer your question?

BR Marius and Espen

UVVM, 2025-09-29:
Any updates on this issue @roynil ?

closed 2026-10-21 by UVVM