UVVM/UVVM #133: AXILITE independent channels

stfarley, 2021-03-05:
2020.10.06 added support for for independent channels. In cases where the target being accessed by the core only supports one channel access at a time each await completions must be added around each access. Perhaps a config parameter for this feature would be better

mariuselv, 2021-03-08:
Hi,
We belive this is outside the standard of the AXI-Lite; could you clearify the use case?

Br,
Marius

stfarley, 2021-03-08:
Before the update the BFM would not issue a Read and Write cycle simultaneously. In our case we have a couple of very simple slaves that do not support this. (Perhaps not strictly AXI4-Lite compliant but they do work with any host that we are using.
The other issue I suppose if testing for 100% AXI-Lite Compliancy, would be that one might want to be  able to check that the client behaves itself with sequential access.
To be clear I can work around this with await completions but need to go through all of my test code and add them.

mariuselv, 2021-03-15:
Hi,
I will create a ticket for adding a configuration option so that the VIP can be run in this mode. 

Br,
Marius

jorgenkrohn, 2021-11-15:
Hi,
This is fixed in v2021.10.22.

A new element in the BFM config record is added. This is called 'force_single_pending_transaction' and can be set to true to require that a transaction is completed before starting the next transaction.

closed 2021-11-15 by UVVM