UVVM/UVVM #141: Maximum for cmd_idx?

roynil, 2021-04-28:
I have a test case that ran for a very long time. I think it might have wrapped cmd_idx. 

```vhdl
# ** Fatal: (vsim-3421) Value -2147483648 is out of range 0 to 2147483647.
#    Time: 245549350 ns  Iteration: 1  Process: /my_vvc/cmd_executor File: UVVM/uvvm_util-2.15.1/src/protected_types_pkg.vhd Line: 188
# Fatal error in Subprogram t_protected_check_counters.decrement at UVVM/uvvm_util-2.15.1/src/protected_types_pkg.vhd line 188
# 
# HDL call sequence:
# Stopped at UVVM/uvvm_util-2.15.1/src/protected_types_pkg.vhd 188 Subprogram t_protected_check_counters.decrement
# called from  UVVM/uvvm_util-2.15.1/src/methods_pkg.vhd 5798 Subprogram check_value_in_range
# called from  UVVM/uvvm_util-2.15.1/src/methods_pkg.vhd 6045 Subprogram check_value_in_range
# called from  UVVM/uvvm_vvc_framework-2.11.1/src/ti_protected_types_pkg.vhd 235 Subprogram t_vvc_activity.priv_report_vvc_activity
# called from  vip_my_vvc/src/vvc_methods_pkg.vhd 612 Subprogram update_vvc_activity_register
# called from  vip_my_vvc/src/my_vvc.vhd 230 Process cmd_executor
# 
#  tb
#  UVVM/uvvm_util-2.15.1/src/protected_types_pkg.vhd 188 return [address 0x7feff09fc20e] Subprogram t_protected_check_counters.decrement
# called from  UVVM/uvvm_util-2.15.1/src/methods_pkg.vhd 5798 return [address 0x7feff0a27d8b] Subprogram check_value_in_range
# called from  UVVM/uvvm_util-2.15.1/src/methods_pkg.vhd 6045 return [address 0x7feff0a2e5c2] Subprogram check_value_in_range
# called from  UVVM/uvvm_vvc_framework-2.11.1/src/ti_protected_types_pkg.vhd 235 return [address 0x7feff0ae6978] Subprogram t_vvc_activity.priv_report_vvc_activity
# called from  vip_my_vvc/src/vvc_methods_pkg.vhd 612 return [address 0x7feff0d0b00e] Subprogram update_vvc_activity_register
# called from  vip_my_vvc/src/pixel_if_vvc.vhd 230 return [address 0x7fef8d2eb44d] Process cmd_executor

mariuselv, 2021-04-28:
Hi,
I think you are right. The vvc_cmd_idx is a natural, hence its maximum value will be 2147483647. First time I've seen anyone encounter this.

Br,
Marius

mariuselv, 2021-04-29:
Hi Ronny,
I've done some checking and the error is triggered by the `priv_check_counters()` method wrapping on line 188 in the protected_types_pkg. We will add a fix that will perform some checks and print out a warning if the counter is about to wrap around - expect the fix to be included in the next release. In the meanwhile you can try to comment out line 180 and 188 in the protected_types_pkg.

Br,
Marius

mariuselv, 2021-04-30:
Hi Ronny,
I've checked and actually this issue is due to a check_counter that wraps around. There is a bug in the check_counter method, but I also think we need to add an internal checker with some sort of alert messaging if the counter is reaching its upper limit. I think we can have this fix in our next release, and in the meanwhile you can make the fix yourself by changing from `-1` to `1` in the `protected_types_pkg`:
`    procedure decrement(
      check_type : t_check_type;
      number     : integer := -1
    ) is
`

to 

`    procedure decrement(
      check_type : t_check_type;
      number     : integer := 1
    ) is
`

Remember to fix both the declaration and body.

Br, 
Marius

mariuselv, 2021-05-26:
Fixed in v2021.05.26 release

closed 2021-05-26 by mariuselv