UVVM/UVVM #144: check_value() fails when using wait_num_rising_edge(clk, 1) instead of wait until rising_edge(clk)

guilbaudl, 2021-05-14:
Hey everyone,

Note: I am using ModelSim INTEL FPGA STARTER EDITION 10.6d Revision: 2018.02 for my simulation.

I am using a simple testbench found on github that is using UVVM. I changed some things to adapt it to my project, and I have found an issue, but I don't know how it can happens. The issue occurs there:

`for il in 0 to 267 loop
      wait_num_rising_edge(clk, 1);
      -- wait until rising_edge(clk);
      check_value((((encodeout_r /= decodein_n_r) and (encodeout_r /= decodein_p_r))), false, tb_warning,  "Check encoding", C_SCOPE);
    end loop;`

All the signals are well defined and assigned just after the wait above.

The issue is that when I am using the `wait_num_rising_edge(clk, 1)` procedure, my test fails
[image]

But you can see in the note that the result is correct. However, when I am using the native `wait until rising_edge(clk)` procedure, I don't have the issue (in fact I have it, but only on the first value tested and it is correct too, so still weird that this can happen).

When I take a look at the procedure `wait_num_rising_edge`, it is define like that:

`-- Wait for number of rising edges
    if num_rising_edge /= 0 then
      for i in 1 to num_rising_edge loop
        wait until rising_edge(clk);
      end loop;
    end if;`

So, I don't get how there can be any trouble. If you need more details about my files, ask me!

Br

mariuselv, 2021-05-14:
Hi @guilbaudl 
I can not see exactly why you get different results. Does the test with await_num_rising_edge() method fail on the first iteration of the loop? And with the regular 'await until rising_edge()' you only have a failing test at the initial loop iteration? As you have already pointed out there are not many changes in behavior from the UVVM method and the regular 'await' statement, but there is a delta cycle wait in the UVVM method. Could you try to add that to your code? I.e.

for il in 0 to 267 loop
  wait until rising_edge(clk);
  wait for 0 ns;
  check_value(((encodeout_r /= decodein_n_r) and (encodeout_r /= decodein_p_r)), false, tb_warning,  "Check encoding", C_SCOPE);
  log('Loop iteration=' & to_string(il));
end loop;

Br, 
Marius

guilbaudl, 2021-05-14:
Hi @mariuselv 

The test with the UVVM method `wait_num_rising_edge()` fails at the the second iteration, but this is the first one after the reset
[image]

With the regular `wait until rising_edge()` there is just this fail at the second iteration.

When adding `wait for 0 ns`, I have the same issue than when I use the UVVM method, you though well. However, when I write `wait for clk_period`, I get the same result than if I just write `wait until rising_edge()`

It seems a bit dark to understand how these errors could occur considering that the values stored in the signals are correct

mariuselv, 2021-05-26:
Hi,
This seems to be a delta cycle problem, i.e. one or more of the encodeX / decodeX are assigned a new value on the same delta cycle as the check - after the rising edge of the clock. Open the simulations in wave view and check when the values get updated.

Br,
Marius

UVVM, 2023-03-30:
Hi,
Did you find a solution to your problem or UVVM?

-- Marius and Espen

closed 2023-10-09 by guilbaudl