UVVM/UVVM #146: VVC / BFM assistance required

KevNug, 2021-05-18:
Hi,

Wrt a simple updn Counter vhdl example trying to implement BFM and VVC to it, seen the following link : 

https://www.edaplayground.com/x/2LhW

Wrt this counter example on EDA Playground platform using  VVC framework & BFM:
It uses a constructor, command interpreter (includes  a state machine process within it), command executor process, command  termination handler etc.

This looks quite complicated to follow eg **counter_vvc.vhd**, counter_bfm_pkg.vhd etc. Is creating a VVC so complex ?

It states at the top of the counter_vvc.vhd that this VVC was generated with Bitvis VVC Generator. Can I generate my counter VVC example using a VVC Generator? If yes, how do I get the VVC Generator installed? Please comment here.

Do you have any other example of eg a simple counter to suggest how to best implement a VVC for it ?

Currently using features (ie procedures) of Util library within my Counter testbench and this looks ok so far BUT having difficulties understanding how to convert this verification platform to using VVC / BFM 

Your prompt reply to this matter will be appreciated.

Kind regards,

Kevin

mariuselv, 2021-05-18:
Hi Kevin, 

There is a vvc_generator.py script in uvvm_vvc_framework/script/vvc_generator. This script will give you all the files needed to create a VVC with a BFM, where code comments tell you where to start writing your code. In general, a BFM package consist of procedures that the VVC (executor process) will call, based on which type of VVC command you send to it. In the VVC methods package you have all the procedure calls that you typically will be using from the testbench sequencer, i.e. spi_write(), spi_read() and so on. The purpose of a VVC is to allow for the testbench sequencer to continue on doing some other work while the VVC is busy executing the command that was issued from the TB sequencer. 

You can have a look at the VVC framework manual in uvvm_vvc_framework/doc folder, and the Bitvis VIP Clock Generator is probably the easiest VVC you can look at to get an understanding of the concept and command flow. We also have an [UVVM forum](https://forum.uvvm.org) where you can post questions and maybe find answers to other questions. 

Best regards,
Marius

KevNug, 2021-05-18:
Hi Marius,

Thanks for your reply. 

The vvc_generator.py looks to be a python script. So I will require installing Python to run it. 

Just briefly looked at the VCC Framework manual.  Section 3.4 'Making your own VVC and VVC methods' . It states that all the required BFM procedures must be available prior to making a VVC. 

>> In general, a BFM package consist of procedures that the VVC (executor process) will call, based on which type of VVC command you send to it

I started creating my own VHDL procedures and placed them eg within ClockCycleCounter_bfm_pkg.vhd. Simulated few of the procedures written and they were ok. So I can carry on creating more procedures within the *_bfm_pkg.vhd and that should be ok prior to making the VVC. Please confirm. 

Which version of Python do you suggest? Is Python 3.0 ok?

When I run the .py script from what I read below within the README.txt file of the Clock Generator the *_bfm_pkg.vhd file is one of the output files after executing .py script. Are there any input files the .py script expects? And if yes, do they have to be located same directory as the .py script is located?

Planning to go through the Bitvis VIP Clock Generator VVC you suggested in more detail soon. Looks like this a Clock Generator VVC generation example. Seen the vvc_generator.py script within the uvvm_vvc_framework\script folder.  Just opened also the README.txt file and it states that only Python 3 is supported, and the script has been tested with version 3.4. So it looks like Python 3.0 or 3.4 should be ok. That README.txt could answer my other questions.

This README.txt file noticed that the following files will be generated (including the *_bfm_pkg.vhd one) :

- <name>_vvc.vhd
- <name>_<channel>_vvc.vhd
- <name>_bfm_pkg.vhd
- vvc_cmd_pkg.vhd
- vvc_methods_pkg.vhd

So, from this, do I then add all my VHDL procedures within the  <name>_bfm_pkg.vhd generated file ? So, it should have code comments within it telling me where to start writing my code. Is this correct? Please re-confirm.

Best regards,
Kevin

KevNug, 2021-05-18:
Hi Marius,

Python 3.4 has reached end-of-life. Python 3.4.10, the final release of the 3.4 series..

Can I install the latest python? ie python 3.9.5 windows 10 ?
Will this work ok with your UVVM scripts ? 
If not, which other Python version do you recommend installing? 3.6 or 3.7 or 3.8 or 3.9 ?

Your prompt reply to this matter will be appreciated

Regards,
Kevin

KevNug, 2021-05-18:
Hi Marius,

I just installed Python 3.9.5 from python.org website and get IDLE Shell 3.9.5 to work ok. I executed the vcc_generator.py script on IDLE Shell 3.9.5 and answered the following questions as : 

 Please enter the VVC Name (e.g. SBI, UART, axilite): SBI
 Generate VVC with extended UVVM features? [y/n]: n
 Set the number of concurrent channels to use [1-99], press enter for default(1): 1
 Shall the VVC have multiple executors? [y/n]: n

This generated VVC within the vvc_generator/output directory

sbi_bfm_pkg.vhd
sbi_vvc.vhd
vvc_cmd_pkg.vhd
vvc_context.vhd
vvc_methods_pkg.vhd

So the above generated code is provided as starting point for building a VVC

Looks like Python 3.9.5 for windows is ok to use. Please confirm.

Regards,

Kevin

mariuselv, 2021-05-19:
Hi Kevin,
Yes, you can use any version of Python 3.0 and above. The generated files in the output directory contain all necessary code and have comments with '<USER-INPUT>' where you need to add code. As a starting point you can start by adding code to the BFM package and call those procedures directly from your TB sequencer. When you get the BFM package working as expected you can start on the VVC and add VVC commands that you will use from the TB sequencer to run the procedures in the BFM package.

Br,
Marius

KevNug, 2021-05-19:
Hi Marius,

Great. Thanks for your replies so far.

Is it ok at the moment with : 
    Generate VVC with extended UVVM features? [y/n]: n

ie no Scoreboard and transaction info.

Please advise here when I will require to enable this option.

Kind regards,
Kevin

KevNug, 2021-05-19:
Hi Marius,

Noticed that the generated files if eg I add a name 

    Please enter the VVC Name (e.g. SBI, UART, axilite): ClockCycleCounter

the generated files just use all lowercase name ie clockcyclecounter_* to the generated files instead of the ClockCycleCounter_* one.

Any suggestions on how to keep the uppercase ie same name inputed ? 

Kind regards,
Kevin

mariuselv, 2021-05-19:
Hi,
It's been a while since I last had look at the script, but I do not think there are any settings to select this.

Br,
Marius

KevNug, 2021-05-19:
Hi Marius,

Ok thanks. Since using purely VHDL no worries at the moment. Will keep the names like that.

If you can also provide a reply to my previous message, I will appreciate. Thanks.

Best Regards,
Kevin

closed 2021-05-20 by mariuselv

KevNug, 2021-05-20:
Before this ticket is closed completely, if you can answer the following questions raised already, I will appreciate:

The testbench template code that contains the test sequencer process is not generated by the python script.
I have to create this by myself. Please confirm.

Is it ok at the moment with :
Generate VVC with extended UVVM features? [y/n]: n

ie no Scoreboard and transaction info.

Please advise here when I will require to enable this option.