UVVM/UVVM #147: check_value() procedure use for a simple counter

KevNug, 2021-05-18:
Hi,

I have a simple counter that is enabled counting up when in_enable = '1' and if reaches MAX_COUNT 536870911 it resets back to 0.

Trying to see how I can use the check_value() overload procedure to do self checking for pass/fail ie is the counter incrementing ok? paused when enable is '0' etc.

Noticed within the methods_pkg.vhd the following check_value() procedure which I will like to call this (using the correct parameters within my testbench to perform this to check if the counter is incrementing ok, paused when enable ='0' and back counting up if enable goes back to '1' and reset to 0 if MAX_COUNT is reached.

Declared already in my testbench :

library uvvm_util;
  context uvvm_util.uvvm_util_context;

& compiled all the necessary packages.

Managed to do the following at the moment : 

Used gen_pulse procedure (using specific overload where pulse always takes  a value of '1') to generate active high reset pulse for eg 10 clk cycles  long.

Used clock_generator overload procedure to generate the clock. It also  checks that the  clock_high_time < clock_period within the procedure. It reports a TB_ERROR  as expected if condition this is not true.

Currently calling the gen_pulse procedure from a p_main test sequence  process within the testbench. The clock_generator procedure is also called  within the testbench architecture part and it is enabled/disabled within  the p_main process.

All works as expected so far.

Having issues trying to use correctly the check_value() procedure as described above within my p_main test sequence testbench. Any help here will be appreciated.

---

   impure function check_value(
    constant value            : t_unsigned_array;
    constant exp              : t_unsigned_array;
    constant match_strictness : t_match_strictness;
    constant alert_level      : t_alert_level;
    constant msg              : string;
    constant scope            : string         := C_TB_SCOPE_DEFAULT;
    constant radix            : t_radix        := HEX_BIN_IF_INVALID;
    constant format           : t_format_zeros := KEEP_LEADING_0;
    constant msg_id           : t_msg_id       := ID_POS_ACK;
    constant msg_id_panel     : t_msg_id_panel := shared_msg_id_panel;
    constant caller_name      : string         := "check_value()";
    constant value_type       : string         := "t_unsigned_array"
    ) return boolean is
      variable v_len_check_ok : boolean := (value'length = exp'length);
      variable v_dir_check_ok : boolean := (value'ascending = exp'ascending);
      -- adjust for array index differences
      variable v_adj_idx      : integer := (value'low - exp'low);
    begin
    protected_check_counters.increment(CHECK_VALUE);

    check_value(v_dir_check_ok = true, TB_WARNING, "array directions do not match", scope);
    check_value(v_len_check_ok = true, TB_ERROR, "array lengths do not match", scope);

    for idx in exp'range loop
      -- do not count CHECK_VALUE multiple times
      protected_check_counters.decrement(CHECK_VALUE);
      if not(check_value(std_logic_vector(value(idx + v_adj_idx)), std_logic_vector(exp(idx)), alert_level, msg, scope, radix, format, msg_id, msg_id_panel, caller_name, value_type)) then
        return false;
      end if;
    end loop;
    return true;
  end;

----

My simple counter vhdl code follows : 
=========================

entity ClockCycleCounter is
port
(
	in_clock	   : in std_logic;
	in_reset	   : in std_logic;
	
	in_enable   : in std_logic;
	out_counter : out Words(1 downto 0)
);
end entity;

architecture main of ClockCycleCounter is

	-- 1073741823 = 111111111111111111111111111111
	constant MAX_COUNT : natural := 536870911 ;
	
	signal counter  : natural range 0 to MAX_COUNT := 0;
	signal std_value     : std_logic_vector(28 downto 0);
	signal full_std_value: std_logic_vector(31 downto 0);

begin
	process(in_clock, in_reset)
	begin
		if(in_reset = '1') then
			counter <= 0;
		elsif (rising_edge(in_clock)) then
			if (in_enable = '1') then
				if (counter = MAX_COUNT) then
					counter <= 0;
				else
					counter <= counter + 1;
				end if;
			end if;
		end if;
	end process;

	std_value <= std_logic_vector(to_unsigned(counter, 29));
	full_std_value <= "000" & std_value;

	out_counter(0) <= full_std_value(15 downto 0);
	out_counter(1) <= full_std_value(31 downto 16);

end architecture;

---

Regards,

Kevin

mariuselv, 2021-05-19:
Hi Kevin,
Make a variable v_prev_count_value : natural := 0, then add the check_value() and the v_prev_count_value increment as follows (I have not tested the code, but hopefully you get the idea):

if (counter = MAX_COUNT) then
    counter <= 0;
    v_prev_count_value := 0;
else
    if counter = 0 then
        check_value(v_prev_count_value, counter, TB_ERROR, "checking initial counter values", C_SCOPE);
    else
        check_value(v_prev_count_value+1, counter, TB_ERROR, "checking counter increment value", C_SCOPE);
        v_prev_count_value := v_prev_count_value + 1;
    end if;
    
    counter <= counter + 1;
end if;

Br,
Marius

KevNug, 2021-05-19:
Hi Marius,

Thanks for your code example. 

So where it is better to place this piece of code? 

Inside a new procedure included within the *_bfm_pkg.vhd? and this will then be called from the testbench p_main test_sequencer process?

Kind regards,
Kevin

mariuselv, 2021-05-19:
Hi,
No, this was just to show you how this would have been implemented inside the TB code you provided. 
Inside a BFM (Bus Functional Model) there should be methods that perform the pin wiggle towards the DUT (device under test), i.e. driving an interface or driving/reading GPIOs. It would probably be easier if you describe what kind of DUT you are trying to verify - or if its just is to get an overview of the system and functionality. Could you move your questioning to the [forum](https://forum.uvvm.org) so that questions and answers will be more accessible for others to see - we close issues on GitHub and then they get sort of "hidden".

Br,
Marius

closed 2021-05-19 by mariuselv

KevNug, 2021-05-20:
Hi Marius,

The code I provided is just a simple counter which is the DUT. 
Trying to familiarize myself with the UVVM uvvm_util procedures (to start with) + VVC

Decided to check_value of the output eg out_counter(0) of the DUT and not the internal counter value signal.

Used eg 

check_value(std_logic_vector(to_unsigned(to_integer(unsigned( v_prev_count_value(15 downto 0) )) + 1, 16)), out_counter(0), TB_ERROR, "checking out_counter(0) increment value", C_SCOPE);

[ Note:  out_counter(0) is 16bit ]

Managed to get the simulation to work (not using VVCs yet)
Resetting the counter, Enabling, Disabling and re-enabling it. I used check_value() at different points to check expected value against the simulated value and there were no failures ie no TB_ERRORs reported. I got few TB-ERRORs to start with but then rectified them.

Thanks for your assistance so far.

Kind regards,
Kevin