UVVM/UVVM #148: UVVM VVC of a simple counter (having issues getting it to simulate with counter_vvc in place)

KevNug, 2021-05-26:
Hi Marius,

Looked at the following link which has a simple counter and tried to follow same approach like they do to build the UVVM platform with VVC in place

         https://www.edaplayground.com/x/2LhW

This simple counter design.vhd, testbench is not complete ie test sequencer process is missing calling the procedural commands etc

I am using a different counter to this one just incrementing and reseting. 

Used gen_pulse within the testbench to create an active high pulse for the reset. This works ok. Reset connects directly to the DUT (counter) within the test harness (ie it is not part of my counter_vvc). This reset pulse works fine.

Instantiated also within my testbech the bitvis  clock_generator_vvc. This works fine too.

I created a BFM procedure counter_count_up and placed this within the counter_bfm_pkg

I called this counter_count_up procedure within my testbench Test Sequencer process and the counting up works fine right after reset.

However, having problems at the moment when I added the BM procedure call in the counter_vvc within the VVC dedicated operations. case v_cmd.operation COUNT_UP was used for this and within vvc_cmd_pkg.vhd I included COUNT_UP within the t_operation type list. The clk is visible but the counter_vvc counter interface 'enable' and 'out_counter' interface signals I cannot see any activity on these lines. Wondering why is this ?

The counter_count_up procedure just generates an active high enable signal (out) and counts up. It also does self checking of the counter at various points using the check_value uvvm utils procedure. 

Your reply to this matter will be appreciated

Regards,

Kevin

KevNug, 2021-05-26:
I have also sent this query to the forum link you sent me ..

mariuselv, 2021-05-27:
Hi Kevin,
Please see my answer on the UVVM forum.

Br,
Marius

closed 2021-05-27 by mariuselv

KevNug, 2021-05-28:
Hi Marium,

Thanks for that info. It was very helpful.

Raised another ticket on the forum a while ago related to vcom warnings:

vcom-1236 - Shared variables must be of a protected type
vcom-1346 - Default expression of interface object is not globally static
vcom-1090 - Possible infinite loop: Process contains no WAIT statement

created a warning.txt file but could not attach the whole .txt doc there. It did not accepted .txt format. Neither .doc was listed

I included a summary list of the warnings using :::: where the warnings were identical in nature

How can I attach the entire .txt file for review? All of these warnings are related to the uvvm templates. Concern about the 2 vcom-1090 ones. 

Can we ignore all of them? 

Kind regards,
Kevin