UVVM/UVVM #150: Recommended method to get result of testbench to calling do_script and eventually the Makefile

ednes, 2021-06-07:
Hi all,

In our project, we have several testbenches. 

We want to run regression tests, based on Makefiles. I can grep for the string "Simulation SUCCESS" in the logfile, but I was wondering if there is a dedicated variable/signal within the testbench available which we can extract using a modelsim.do file

e.g. my modelsim.do file now contains:
```
run 100 ms
set UVVM_STATUS [examine /global_signals_and_shared_variables/shared_uvvm_status/found_unexpected_error_or_worse]
if {$UVVM_STATUS == 0} {exit -code 0} else {exit -code -1}
```
The exit status of the modelsim.do file is catched in the makefile.

I found uvvm_status record, but that is initalized with "no errors" so if I check that before the entire simulation is done, I'm missing that the simulation has not completed.

Would you be willing to accept a pull-request, which adds a "simulation_status" integer to this record, which is initalized as -2 (simulation still busy), gets the value -1 (error) when the simulation ends with any warning or error, and gets the value 0 (all is fine) when the simulation ends with "all succeeded"?

Thanks for making UVVM available!

Edwin Hakkennes

mariuselv, 2021-06-07:
Hi Edwin,
Can you use the shared_uvvm_status from the global_signals_and_shared_variables_pkg:
`shared variable shared_uvvm_status             : t_uvvm_status := C_UVVM_STATUS_DEFAULT;`

It is a record type where you have the following information available:
`  type t_uvvm_status is record
    found_unexpected_simulation_warnings_or_worse     : natural range 0 to 1; -- simulation end status: 0=no unexpected, 1=unexpected
    found_unexpected_simulation_errors_or_worse       : natural range 0 to 1; -- simulation end status: 0=no unexpected, 1=unexpected
    mismatch_on_expected_simulation_warnings_or_worse : natural range 0 to 1; -- simulation status: 0=no mismatch, 1=mismatch
    mismatch_on_expected_simulation_errors_or_worse   : natural range 0 to 1; -- simulation status: 0=no mismatch, 1=mismatch
    info_on_finishing_await_any_completion            : t_info_on_finishing_await_any_completion; -- await_any_completion() trigger identifyer
  end record t_uvvm_status;`

Br,
Marius

ednes, 2021-06-07:
Hi Marius,

I was hoping on a single integer to be read. I'm not sure, how to parse the info_on_finishing_await_any_completion. This is yet another structure to dive into.

My suggestion is to add one single integer to this record, which encapsulates all information needed for a regression test in just that one integer.

```
simulation_status: integer range -2 to 0; -- simulation status: -2=simulation still running, -1=simulation ended unsuccesfull, 0=simulation ended succesfull
```

Or am I missing the meaning of the four variables already present?

I currently check all 4 to be zero, as well as the shared_log_hdr_for_waveview. See my .do file below.

```
onerror resume
run -a

quietly set found_unexpected_simulation_warnings_or_worse     [examine -radix decimal /global_signals_and_shared_variables_pkg/shared_uvvm_status.found_unexpected_simulation_warnings_or_worse]
quietly set found_unexpected_simulation_errors_or_worse       [examine -radix decimal /global_signals_and_shared_variables_pkg/shared_uvvm_status.found_unexpected_simulation_errors_or_worse]
quietly set mismatch_on_expected_simulation_warnings_or_worse [examine -radix decimal /global_signals_and_shared_variables_pkg/shared_uvvm_status.mismatch_on_expected_simulation_warnings_or_worse]
quietly set mismatch_on_expected_simulation_errors_or_worse   [examine -radix decimal /global_signals_and_shared_variables_pkg/shared_uvvm_status.mismatch_on_expected_simulation_errors_or_worse]
quietly set shared_log_hdr_for_waveview                       [examine /global_signals_and_shared_variables_pkg/shared_log_hdr_for_waveview]
quietly set shared_log_hdr_short                              [string range $shared_log_hdr_for_waveview 1 20]

quietly if {$shared_log_hdr_short == "SIMULATION COMPLETED" &&
            $found_unexpected_simulation_warnings_or_worse == 0 &&
            $found_unexpected_simulation_errors_or_worse ==0 &&
            $mismatch_on_expected_simulation_warnings_or_worse== 0 &&
            $mismatch_on_expected_simulation_errors_or_worse == 0} {
       set simulation_ok 1
    } else {
       set simulation_ok 0}

echo "Simulation ok =" $simulation_ok

if {$simulation_ok ==1 } {
  echo "TEST PASSED"
  if [batch_mode] {
    exit -code 0
  }
} else {
  echo "TEST FAILED"
  if [batch_mode] {
    exit -code 1
  }
}
```

I would love to replace all statements starting with `quietly` with just reading the one new variable. Which gets it -2 value upon initializing, and the -1 or 0 value at the time the final pass/fail message is printed in the log.

Best regards,

Edwin

mariuselv, 2021-06-07:
Hi,
We had a somewhat similar question a while a go (https://github.com/UVVM/UVVM/issues/82) where we added a change so the simulator would exit with a non-zero exit code when the alert counter exceeds the expected number of alerts. Is this a feature you could utilize? 

Br,
Marius

ednes, 2021-06-07:
Hi,

That should be also usable. I will give this a go.

Br,

Edwin

levi-ee, 2023-05-26:
Hi all,

Unfortunately, Modelsim seams to ignore the return status of the std.env.stop(X) command (unlike GHDL), so #82 didn't work for me.
TLDR if someone is still looking for a final answer to this question.

So, , this is what I did (based on previous ): 

```tcl
vsim work.tb_test 
run -all
set unexpected_errors [examine -radix decimal /global_signals_and_shared_variables_pkg/shared_uvvm_status.mismatch_on_expected_simulation_errors_or_worse]
exit -f -code $unexpected_errors
```

UVVM, 2026-10-27:
A new field has been added `shared_uvvm_status.simulation_status : natural range 0 to 2`:

- 2 = not completed. This is the default, and it stays 2 until report_alert_counters(FINAL) is called.
- 1 = failed. This uses the same conclusion as the printed "Simulation SUCCESS/FAILED" line, i.e. no mismatch on ERROR/TB_ERROR/FAILURE/TB_FAILURE alerts. Warnings remain visible in the existing *_warnings_or_worse fields
- 0 = passed

This has been released in v2 2026.10.27

closed 2026-10-27 by UVVM