UVVM/UVVM #152: AXI4Stream BFM questions

ednes, 2021-07-01:
Hi, I'm writing a simple testcase connecting an AXI4stream master to a slave. As of now, nothing in between the two, the user-logic will follow once the stream is "flowing" as intended. 

1) The code is complaining that _bytes methods are deprecated, but the record between the BFM and the VVC can only transport bytes, and therefore the vvc executes the transfer using the _bytes functions. Somewhere in the UVVM code, my t_slv_array is mapped to a t_byte_array. As the bus is 16 bit, two bytes are driven on the bus in one cycle.

Is there a simple example on how to simulate this for a 16 bit bus? I tried different options for byte_endianness, and now the data is correct on the bus, but the logging is still about the bytes transferred, not the words. How can I fix that?

2) I want to implement a stream bus without back-pressure, driving ready to the master as a constant one. However, the slave BFM starts to assert ready only after it detected valid. Is there a way to instruct the slave BFM to "always be ready to receive data"?

I tried `ready_default_value     => '1' ` in the config-word of the slave, as well as , but that doesn't seem to make a difference.

Thanks for your reply,

Edwin

[image]

ednes, 2021-07-05:
Hi all,

To clarify: I want to transfer 16 bit (and in other places 32 bit) values over the axi-stream-bus. I want to be able to send and receive them using somthing in the line of:

```
constant c_data_to_transmit : t_slv_array(0 to 2)(15 downto 0) := (x"1234", x"5678", x"9abc");

begin testcase
  i_test_harness : entity lib_tb_adc_convert.th_adc_convert;

  p_stim : process is
  begin  -- process p_stim
    -- Wait for UVVM to finish initialization
    await_uvvm_initialization(VOID);

    shared_axilite_vvc_config(1).bfm_config.clock_period := 8 ns;

    wait for 2 ns;
    -- start clock
    start_clock(CLOCK_GENERATOR_VVCT, 1, "Start clock generator");

    wait for 1 us;
    axistream_transmit(AXISTREAM_VVCT, 2, c_data_to_transmit(0 to 2), "Send a packet to DUT");

    wait for 200 ns;

    axistream_expect(AXISTREAM_VVCT, 3, c_data_to_transmit(0 to 2), "Receive and check packet");

    wait for 1 us;

```
And I want the data on the bus to be "1234" not "3421". 

The current logging is: 
``` 
# UVVM: ID_PACKET_DATA                  1160.0 ns  AXISTREAM_VVC,2                axistream_transmit(6B)=> Tx x"34", byte# 0. 'Send a packet to DUT'  [3]
# UVVM: ID_PACKET_DATA                  1160.0 ns  AXISTREAM_VVC,2                axistream_transmit(6B)=> Tx x"12", byte# 1. 'Send a packet to DUT'  [3]
```

I would prefer this as:
```
# UVVM: ID_PACKET_DATA                  1160.0 ns  AXISTREAM_VVC,2                axistream_transmit(3W)=> Tx x"1234", Word #0. 'Send a packet to DUT'  [3]
```

The same holds for the receiving messages.

Setting the byte_endianness in the configuration to either LOWER_BYTE_LEFT, LOWER_BYTE_RIGHT, LOWER_WORD_LEFT, LOWER_WORD_RIGHT, FIRST_BYTE_LEFT or FIRST_BYTE_RIGHT doesn't seem te make a difference.

I can't imagine I'm the first to struggle with this...

Kind regards,

Edwin

erick166, 2021-07-06:
Hi Edwin,

1. The deprecated messages are printed so that the users avoid using the BFM _bytes methods, even though at the moment they are still used as base procedures by the VVC. This will be fixed in the next major release due to backwards-compatibility. In the meanwhile, you can use **disable_log_msg(ID_UTIL_SETUP)** to hide the log messages.

1.1 The log always prints each byte separately because the base procedures work on a byte level, however you can see in the timestamp that both bytes in a 16-bit vector are sent at the same moment, e.g. byte#0 and byte#1 at 105.0 ns

# UVVM: ID_PACKET_DATA                   105.0 ns  AXISTREAM_VVC,2                axistream_transmit(6B)=> Tx x"12", byte# 0. 'Send a packet to DUT'  [5]
# UVVM: ID_PACKET_DATA                   105.0 ns  AXISTREAM_VVC,2                axistream_transmit(6B)=> Tx x"34", byte# 1. 'Send a packet to DUT'  [5]
# UVVM: ID_PACKET_DATA                   122.5 ns  AXISTREAM_VVC,2                axistream_transmit(6B)=> Tx x"56", byte# 2. 'Send a packet to DUT'  [5]
# UVVM: ID_PACKET_DATA                   122.5 ns  AXISTREAM_VVC,2                axistream_transmit(6B)=> Tx x"78", byte# 3. 'Send a packet to DUT'  [5]
# UVVM: ID_PACKET_DATA                   132.5 ns  AXISTREAM_VVC,2                axistream_transmit(6B)=> Tx x"9A", byte# 4. 'Send a packet to DUT'  [5]
# UVVM: ID_PACKET_DATA                   132.5 ns  AXISTREAM_VVC,2                axistream_transmit(6B)=> Tx x"BC", byte# 5. 'Send a packet to DUT'  [5]

This is of course not ideal and the log messages will also be updated to be more like what you suggest when the deprecated methods are removed.

By default the byte_endianness is LOWER_BYTE_LEFT so your example actually prints what I just copied above when you don't configure anything. By the way, the example uses the wrong VVC config **shared_axilite_vvc_config**.

2. Configuring ready_default_value => '1' on the slave VVC should work. Could you show me an example of that part of the code?

Best regards,
Erick

ednes, 2021-07-06:
Hi Erick, 

Thanks for your reply. I mixed up two testcases in my example. The "wrong VVC config shared_axilite_vvc_config" was copied from another testbench.

Apparently, I'm no longer able to get the data in the wrong order on the bus. That's good. Is there any way to use the non-byte method, by changing the UVVM sources myself? What I find strange, is that each of the six methods of endianness on the master, result in the same data, and changing it on the master and not the slave does not make the testcase fail.

As for point 2: I already tried the configuration you suggested. Doesn't make any difference. This is shown clearly in the waveform.

I have added my test harness and testbench to this post. Perhaps, they can be added to the project as example and starting point for other users. After a thorough review and removal of my errors, that is..

Kind regards,

Edwin

[lib_tb_axi_stream.zip](https://github.com/UVVM/UVVM/files/6770932/lib_tb_axi_stream.zip)

erick166, 2021-07-06:
Basically the axistream_transmit() and other "non-byte" procedures are just overloads for the _byte ones, I believe this was done mainly to change the input data parameter from t_byte_array to t_slv_array for more flexibility. But like I mentioned if you disable the log messages with ID_UTIL_SETUP you shouldn't have any issues, unless there's another reason?

I checked your test harness and the problem is the way you are configuring the VVCs. The correct way is to either set the static config in the VVC generic **GC_AXISTREAM_BFM_CONFIG** or using the **shared_axistream_vvc_config(vvc_idx).bfm_config** in your testbench.

There are currently no simple examples other than what is included in the QuickReference, but you can also check the repo UVVM_SUPPLEMENTARY, there you can find all the internal testbenches used for the different libraries, perhaps you can find other useful things there as well.

Best regards,
Erick

ednes, 2021-07-19:
Ok, thanks for pointing to the SUPPLEMENTARY archive. Proved to be very helpfull.

Kind regards,

Edwin

closed 2021-07-19 by ednes