UVVM/UVVM #155: adding "native memory interface" support to SBI

jeras, 2021-07-13:
SBI is similar to native synchronous memory interfaces for FPGA and ASIC memories.
See XPM_MEMORY_SPRAM or similar in
https://www.xilinx.com/support/documentation/sw_manuals/xilinx2021_1/ug974-vivado-ultrascale-libraries.pdf

The main property of this interfaces is, read data is delayed by one clock cycle from chip select `cs`/`ena`.
Some memories can have a higher and/or programmable read delay.
All memory interfaces support back to back accesses, and they can be any mix of read and write accesses,
all accesses take a single clock cycle.
Meaning `cs` is active for a single clock cycle for each access, and in a back to back sequence there are no gaps.
This can be seen as a single stage read pipeline, something similar is used in the AMBA AHB standard.

The SBI VIP is currently unable to perform such accesses.
I tried to configure fixed read timing and with ready wired to '1':
```VHDL
      shared_sbi_vvc_config(1).bfm_config.use_fixed_wait_cycles_read := true;
      shared_sbi_vvc_config(1).bfm_config.fixed_wait_cycles_read     := 1;
```
But it did not work as I expected, read access still took 2 clock cycles.

This protocol can also support back-pressure with `ready`.
Back-pressure is useful when there is an arbiter allowing multiple masters to access the same memory.
The arbiter returns an active ready signal only to the master winning the arbitration.
This setup still allows for full throughput at the RAM.
Read data is available one clock period after `cs and ready = '1'`.

While VVC should be able to achieve full throughput, a BFM would insert an inactive `cs='0'` clock period while waiting for the read result.

This support could probably be added to SBI in a backward compatible way by adding a configuration option similar to `use_fixed_wait_cycles_read` but this one would not disable checking `ready`.

mariuselv, 2021-08-02:
Hi,
This seems like a good idea. Could you implement this as a pull-request?

Br,
Marius

UVVM, 2023-03-30:
Hi,
Could you provide a pull request for this?

-- Marius and Espen

UVVM, 2025-09-29:
Any updates on this issue @jeras ?

jeras, 2025-09-29:
Sorry no, I have not been using UVVM much, and I have no plans to implement this.