UVVM/UVVM #156: Is is possible to create a back to back write sequence with AXI4-Lite VVC

jeras, 2021-07-14:
I have tried to create a sequence of back to back writes, but the VVC inserts idle cycles between writes.
It appears VVC waits for write response to complete before starting a new read address cycle.
I did not read the relevant source code yet.

erick166, 2021-07-22:
I believe the AXI-lite protocol only allows for a burst length of 1 and is required to wait for the write response. Have you tried writing bursts of data with the AXI VVC?

jeras, 2021-07-23:
The AXI4/5-Lite protocol does not support bursts, but it does support pipelining. Multiple write address requests can be issued back to back, and the master can have a pipeline waiting for multiple responses.
While most RTL masters probably do not support access pipelining, a VIP master should.
Otherwise tests for an AXI4-Lite slave with pipelining support do not cover the entire protocol.

I have not tried bursts with the AXI-VCC, since I wrote a Lite slave which does not support bursts.

I attached an example of back-to-back access using Xilinx AXI VIP, I had to set the write pipeline to 2 and the read pipeline to 3, since I have an extra read access delay.
[image]

I still have to add some timing randomization:
- delay between `awvalid` and `wvalid`,
- idle cycles between `awvalid`,
- `bready` and `rready` randomization.

This functionality is also missing from UVVM.

jorgenkrohn, 2021-08-02:
The AXI-Lite VVC supports pipelining and will not wait for the previous transaction to finish before starting the next one. By default, the bfm_config (C_AXILITE_BFM_CONFIG_DEFAULT) adds a clock cycle between transactions. This can be changed by changing the BFM-config in the test sequencer. e.g. 
```
shared_axilite_vvc_config(1).bfm_config.num_aw_pipe_stages := 0;
shared_axilite_vvc_config(1).bfm_config.num_w_pipe_stages := 0;
shared_axilite_vvc_config(1).bfm_config.num_ar_pipe_stages := 0;
shared_axilite_vvc_config(1).bfm_config.num_r_pipe_stages := 0;
shared_axilite_vvc_config(1).bfm_config.num_b_pipe_stages := 0;
```

UVVM, 2024-10-31:
Issue solved with comment on configuration. Will open a new issue on randomisation of ready.

closed 2024-10-31 by UVVM