UVVM/UVVM #157: shared_vvc_activity_register

roynil, 2021-07-15:
I have a testcase with 4 VVC. 3 of them executing some bfm-procedure (that could take 1 to many clock cycles) and one executing insert_delay(1 clock cycle).

I have noticed that when using:
await_completion(ANY_OF, list of all 4 VVCs,..);

It is always the VVC executing insert_delay(1) that triggers the completion regardless if any of the other VVCs finishes.
If I then check the shared_vvc_activity_register the VVCs that executed bfm are still in ACTIVE mode (regardless of if they finished the bfm-procedure). If I follow up the await_completion with a:
wait for 0 ns;
the shared_vvc_activity_register shows the correct status of the other VVCs.

erick166, 2021-07-21:
I'm not sure I understand the problem, using _await_completion(ANY_OF, list of all 4 VVCs,..)_ should wait until the first of the 4 VVCs is finished.

roynil, 2021-07-21:
But if 2 of the VVCs finish the same clock-cycle shouldn't that be reflected in the shared_vvc_activity_register?

erick166, 2021-07-21:
Yes, but the difference in delta cycles could be due to some synchronization of the interface in the VVC when finishing a command. The VVC using a BFM command would end with "wait_on_bfm_exit()" and the VVC with a delay would just wait for the specified time. I would probably need to try the example myself to see what is actually happening.

Is this causing your testcase to fail?

UVVM, 2025-09-29:
Any updates on this issue @roynil ?

closed 2026-10-21 by UVVM