UVVM/UVVM #159: Error: Subtype of "axilite_vvc_master_if" is not fully constrained.

AlbertHoevenaars, 2021-07-22:
Getting this error in my small UVVM testbench. Don't understand why. Does someone have a solution?

`
--------------------------------------------------------------------------------
-- History
--
--     Date
-- (YYYY/MM/DD) Author          Rev.  Change Description
--  ==========  ==============  ====  ==========================================
--  2021/00/00  320123287  0.0   generated file (HDL Designer design)
--
--------------------------------------------------------------------------------
-- Design Description
--
--------------------------------------------------------------------------------
LIBRARY ieee;
  USE ieee.std_logic_1164.ALL;
  USE ieee.numeric_std.ALL;

ENTITY uvvm_axi4dut_tb IS
-- Declarations

END ENTITY uvvm_axi4dut_tb ;
-------------------------------------------------------------------------------
--   generated file (HDL Designer Block Diagram)
--------------------------------------------------------------------------------
LIBRARY ieee;
  USE ieee.std_logic_1164.ALL;
  USE ieee.numeric_std.ALL;

--CONTEXT bitvis_vip_axilite.vvc_context;
LIBRARY bitvis_vip_axilite;
  USE bitvis_vip_axilite.axilite_bfm_pkg.all;
  USE bitvis_vip_axilite.transaction_pkg.all;
 USE bitvis_vip_axilite.axilite_channel_handler_pkg.all;
 USE bitvis_vip_axilite.vvc_methods_pkg.all;
 USE bitvis_vip_axilite.vvc_cmd_pkg.all;

LIBRARY uvvm_util;
  USE uvvm_util. types_pkg.all;
CONTEXT uvvm_util.uvvm_util_context;

LIBRARY uvvm_vvc_framework;
USE uvvm_vvc_framework.ti_vvc_framework_support_pkg.all;

LIBRARY Axi4dutlib;
LIBRARY Axi4dutlib_tb;
LIBRARY bitvis_vip_clock_generator;

ARCHITECTURE struct OF uvvm_axi4dut_tb IS

  -- Architecture declarations
  CONSTANT GC_ADDR_WIDTH : natural :=32;
  CONSTANT GC_DATA_WIDTH : natural := 32;

  -- Internal signal declarations
  SIGNAL I2F_WaitIn            : std_logic                    := '0';
  SIGNAL ImageIn               : tFullInternalFormat4         := zero_tFullInternalFormat4;
  SIGNAL ImageOut              : tFullInternalFormat4;
  SIGNAL RndW8Tst              : std_logic_vector(0 DOWNTO 0) := "0";
  SIGNAL WaitOut               : std_logic;
  SIGNAL axi4Clk               : std_logic;
  SIGNAL axi4Rst               : std_logic;
  SIGNAL axiM2s                : t_axi4l_m2s;
  SIGNAL axiS2m                : t_axi4l_s2m;
  SIGNAL axilite_vvc_master_if : t_axilite_if                 := init_axilite_if_signals(32, 32);
  SIGNAL dout_dut             : std_logic_vector(31 DOWNTO 0);
  SIGNAL empty                 : std_logic;
  SIGNAL ipShellDbg            : std_logic_vector(31 DOWNTO 0);
  SIGNAL ip_clk                : std_logic;
[... 157 lines cut ...]
      GC_RESULT_QUEUE_COUNT_THRESHOLD          => 950,
      GC_RESULT_QUEUE_COUNT_THRESHOLD_SEVERITY => warning
    )
    PORT MAP (
      clk => ip_clk
    );
  i_clock_generator_vvc1 : clock_generator_vvc
    GENERIC MAP (
      GC_INSTANCE_IDX                          => 1,
      GC_CLOCK_NAME                            => "clk",
      GC_CLOCK_PERIOD                          => 10 ns,
      GC_CLOCK_HIGH_TIME                       => 5 ns,
      GC_CMD_QUEUE_COUNT_MAX                   => 1000,
      GC_CMD_QUEUE_COUNT_THRESHOLD             => 950,
      GC_CMD_QUEUE_COUNT_THRESHOLD_SEVERITY    => WARNING,
      GC_RESULT_QUEUE_COUNT_MAX                => 1000,
      GC_RESULT_QUEUE_COUNT_THRESHOLD          => 950,
      GC_RESULT_QUEUE_COUNT_THRESHOLD_SEVERITY => warning
    )
    PORT MAP (
      clk => axi4Clk
    );

END ARCHITECTURE struct;
`

erick166, 2021-07-23:
When you declare the **axilite_vvc_master_if** you need to constrain it, e.g.

```
signal axilite_vvc_master_if : t_axilite_if( write_address_channel(awaddr(C_ADDR_WIDTH-1 downto 0)),
                                             write_data_channel(wdata(C_DATA_WIDTH-1 downto 0),
                                             wstrb((C_DATA_WIDTH/8)-1 downto 0)),
                                             read_address_channel(araddr(C_ADDR_WIDTH-1 downto 0)),
                                             read_data_channel(rdata(C_DATA_WIDTH-1 downto 0)));
```

You can find some information about it in the Quick Reference section 7.

AlbertHoevenaars, 2021-07-27:
erick166, thanks for your comment. 

On the toplevel the following definition can be found "SIGNAL axilite_vvc_master_if : t_axilite_if := init_axilite_if_signals(32, 32);"
So this looks constrained.

I don't understand what's wrong.
Can you give me another hint?

erick166, 2021-07-27:
The function _init_axilite_if_signals(32,32)_ will return a constrained type, however the signal **axilite_vvc_master_if** is still unconstrained. You can try:

```
signal axilite_vvc_master_if : t_axilite_if( write_address_channel(awaddr(C_ADDR_WIDTH-1 downto 0)),
                                             write_data_channel(wdata(C_DATA_WIDTH-1 downto 0),
                                             wstrb((C_DATA_WIDTH/8)-1 downto 0)),
                                             read_address_channel(araddr(C_ADDR_WIDTH-1 downto 0)),
                                             read_data_channel(rdata(C_DATA_WIDTH-1 downto 0))) := init_axilite_if_signals(C_ADDR_WIDTH, C_DATA_WIDTH);
```

AlbertHoevenaars, 2021-07-27:
Erick166, Thanks again.
I managed to get this problem solved. 
Now I get an error for the identifier "C_VVC_CMD_DATA_MAX_LENGTH" which is defined in two packages. bitvis_vip_axilite/transaction_pkg.vhd and in bitvis_vip_clock_generator/vvc_cmd_pkg.vhd. Why is this identifier placed in both package files? Removing it from the clock_generator/vvc_cmd_pkg.vhd give me a lot of other errors. Should I overwrite this value in the generics from the axilite_vvc block.  Documentation seems to be very poor on this.

erick166, 2021-07-27:
That's great 👍 

The constant C_VVC_CMD_DATA_MAX_LENGTH is specific for each VIP library, they share the name because their purpose is the same, usually they are not referenced in the testbench.
You could try using the complete path of the constant in the generics of the axilite_vvc block, e.g. **bitvis_vip_axilite.transaction_pkg.C_VVC_CMD_DATA_MAX_LENGTH** or remove the component declaration of the VVCs and leave only the instantiation, unless there is a special reason why you need to declare the components in your testbench.

AlbertHoevenaars, 2021-07-27:
erick166,

tried several solutions but did not manage to get it done in a nice way. I did not want to touch the original bitvis files but finally after a lot of frustration I changed the original axilite_vvc file. Set the range for ADDR_WIDTH and DATA_WIDTH to a fixed value of 32. When I now compiles I get the error message during the Questasim VCOM command that some pkg files are changed. Very strange. "Recompile work.axilite_vvc because work.vvc_methods_pkg, work.vvc_cmd_pkg have changed."

Some articles and some powerpoints tell me that I could create a UVVM testbench in half a hour. Well, I am an experienced VHDL programmer with more then 20 years experience, but I am already working for 3 days to get a simple testbench running. Terrible.
Sorry to bother you with my frustration.

AlbertHoevenaars, 2021-07-27:
Found out that vvc_methods_pkg is in the clock and in the axi library. Two the same named packages files! The same counts for the vvc_cm_pkg file. Do I only need one file? BTW I compile by  means of a batch script and I use one work library. Is this correct?
`
Rem Build automatic test with Questasim
rem ==================== define PC name
rem
set PROJECT_HOME=%CD%
set FDCSIB_USER_HOME=%PROJECT_HOME%\..

IF %COMPUTERNAME% == XXXX goto PC_ALBERT

GOTO ERROR

rem ============== define paths to tools for every PC
rem
:PC_ALBERT
rem
set QUESTASIM_EXE=C:\pldtools\questasim64_2020.2\win64
set XILINX_VENDOR_LIBS=C:\pld\Vendor_libs\questasim64_2020.2\vivado_2020.2
rem
GOTO START

:START
set  TEST_FILES=%FDCSIB_USER_HOME%\Questasim_batch\Special_Test_files
set  WORK_DIR=%FDCSIB_USER_HOME%\Questasim_batch
Rem
Rem Make new work diretories
Rem
if not exist work goto skip_work
rmdir /Q /S work
:skip_work
Rem
Rem Creat physical directories
Rem 
md work
cd work
rem md axi4adpclib
rem md axi4adpclib_tb
rem md uvvm_util
rem md uvvm_vvc_framework
rem md bitvis_vip_scoreboard
rem md bitvis_vip_axilite
rem md bitvis_vip_clock_generator
rem md ipshelllib
rem md axi4componentslib
rem md flashlitelib
rem md work
rem 
cd..
rem
rem  Create work libraries. Every library will gets its own directory
rem
rem %QUESTASIM_EXE%/vlib -type directory %WORK_DIR%\work\axi4adpclib\work
rem 
rem %QUESTASIM_EXE%/vlib %WORK_DIR%\work\axi4adpclib\work
rem %QUESTASIM_EXE%/vlib %WORK_DIR%\work\axi4adpclib_tb\work
rem %QUESTASIM_EXE%/vlib %WORK_DIR%\work\uvvm_util\work
rem %QUESTASIM_EXE%/vlib %WORK_DIR%\work\uvvm_vvc_framework\work
rem %QUESTASIM_EXE%/vlib %WORK_DIR%\work\bitvis_vip_scoreboard\work
rem %QUESTASIM_EXE%/vlib %WORK_DIR%\work\bitvis_vip_axilite\work
rem %QUESTASIM_EXE%/vlib %WORK_DIR%\work\bitvis_vip_clock_generator\work
rem %QUESTASIM_EXE%/vlib %WORK_DIR%\work\ipshelllib\work
rem %QUESTASIM_EXE%/vlib %WORK_DIR%\work\axi4componentslib\work
rem %QUESTASIM_EXE%/vlib %WORK_DIR%\work\flashlitelib\work
%QUESTASIM_EXE%/vlib %WORK_DIR%\work
rem
rem %MODELSIM_EXE%/vlib %WORK_DIR%/work/nios_model/work
rem
rem  Map libraries to directories
rem 
rem %MODELSIM_EXE%/vmap work          %WORK_DIR%/work
[... 123 lines cut ...]
%QUESTASIM_EXE%/vcom -2008 -work work -cover sbcef -coverNoSub %CD%/../fpga1/ipshelllib/ipshelllib/hdl/ipshell_databuf_pkg.vhd 
%QUESTASIM_EXE%/vcom -2008 -work work -cover sbcef -coverNoSub %CD%/../fpga1/ipshelllib/ipshelllib/hdl/ipshell_trlinsert_pkg.vhd
%QUESTASIM_EXE%/vcom -2008 -work work -cover sbcef -coverNoSub %CD%/../fpga1/ipshelllib/ipshelllib/hdl/ipshell_control.vhd 
%QUESTASIM_EXE%/vcom -2008 -work work -cover sbcef -coverNoSub %CD%/../fpga1/ipshelllib/ipshelllib/hdl/ipshell_front.vhd 
%QUESTASIM_EXE%/vcom -2008 -work work -cover sbcef -coverNoSub %CD%/../fpga1/ipshelllib/ipshelllib/hdl/ipshell_mux.vhd 
%QUESTASIM_EXE%/vcom -2008 -work work -cover sbcef -coverNoSub %CD%/../fpga1/ipshelllib/ipshelllib/hdl/ipshell_mapperctrl.vhd 
%QUESTASIM_EXE%/vcom -2008 -work work -cover sbcef -coverNoSub %CD%/../fpga1/ipshelllib/ipshelllib/hdl/ipshell_top_struct.vhd 
rem
rem Now the files on the toplevel
rem
%QUESTASIM_EXE%/vcom -2008 -work work -cover sbcef -coverNoSub %CD%/../fpga1/axi4adpclib/axi4adpclib/hdl/axi4adpc_regs_pkg.vhd 
%QUESTASIM_EXE%/vcom -2008 -work work -cover sbcef -coverNoSub %CD%/../fpga1/axi4adpclib/axi4adpclib/hdl/axi4adpc_addr_pkg.vhd 
rem                                                            
%QUESTASIM_EXE%/vcom -2008 -work work -cover sbcef -coverNoSub %CD%/../fpga1/axi4adpclib/axi4adpclib/hdl/axi4adpc_prm.vhd 
%QUESTASIM_EXE%/vcom -2008 -work work -cover sbcef -coverNoSub %CD%/../fpga1/axi4adpclib/axi4adpclib/hdl/axi4adpc_iif.vhd 
rem                              
%QUESTASIM_EXE%/vcom -2008 -work work -cover sbcef -coverNoSub %CD%/../fpga1/axi4adpclib/axi4adpclib/hdl/axi4adpc_top_struct.vhd
rem                             
rem testbench files              
rem                              
%QUESTASIM_EXE%/vcom -2008 -work work -cover sbcef -coverNoSub %CD%/../fpga1/axi4adpclib/axi4adpclib_tb/hdl/adpc_stimul.vhd
%QUESTASIM_EXE%/vcom -2008 -work work -cover sbcef -coverNoSub %CD%/../fpga1/axi4adpclib/axi4adpclib_tb/hdl/axi_splitter.vhd
%QUESTASIM_EXE%/vcom -2008 -work work -cover sbcef -coverNoSub %CD%/../fpga1/axi4adpclib/axi4adpclib_tb/hdl/uvvm_axi4adpc_tb_struct.vhd
pause
`

erick166, 2021-07-27:
Sorry that you feel frustrated, it should be easy enough to start using the VVCs by reading the Quick Reference document, perhaps it needs better examples.

Just to be clear, neither of the 2 solutions I mentioned worked for you?
1. Remove the component declaration in the testbench
2. Use the full path of the constant: bitvis_vip_axilite.transaction_pkg.C_VVC_CMD_DATA_MAX_LENGTH 

You might also try looking at the UVVM supplementary repository, there you can find the testbenches for every library and perhaps some useful examples.

Yes, each VIP library has files with common names: vvc_methods_pkg, vvc_cmd_pkg, etc. However, they must be compiled into different libraries so that there's no conflicts between them. This is explained in the VVC QuickRef Section 9.
If you have a look at the compile_order.txt file under bitvis_vip_axilite\script you can see the files needed for that specific library. You can find the same compile_order.txt file in every VIP library.

I hope this helps.

AlbertHoevenaars, 2021-07-28:
erick166, Yes I managed to get things running in Questasim. Checked the order of compilation and the work libraries. Found some errors and after correcting them I has zero compilation errors. 
But now (of coarse) the next problem arise. When I run my testbench in Questasim (64 bit version 2020.04) the design seems to get stuck somewhere. Message = "# ** Error (suppressible): (vsim-3601) Iteration limit 10000000 reached at time 0 ps." It is not in my sequencer, because if I remove all the commands this loop still exist. The DUT I am testing worked already in a plain VHDL testbench. Must be somewhere in the UVVM blocks. 
Any idea or suggestion?

erick166, 2021-07-28:
It's good that you managed to compile without errors.

I can see in your testbench file that there are two clock generator VVCs with the same instance index, the indexes should be unique among VVCs of the same type, try changing one of them. Otherwise there's nothing else that looks strange in the testbench file that you copied here. Remember also to use **await_uvvm_initialization(VOID);** at the beginning of your sequencer.

AlbertHoevenaars, 2021-07-28:
erick166, I changed the number of the GC_INSTANCE_IDX indexes already in the last days. Every uvvm block has a different index. Also the statement await_uvvm_initialization(VOID); is in my seqeuncer.

erick166, 2021-07-28:
Have you instantiated the following in your testbench?
i_ti_uvvm_engine : entity uvvm_vvc_framework.ti_uvvm_engine;

AlbertHoevenaars, 2021-09-01:
erick166, sorry for my very very late replay. Due to some other work and some holidays it took a while. After adding the block "ti_uvvm_engine" everything started. Thanks for that remark! Simulation is now running, although I get an error message on some AxiLite read actions, Looks like if they are coming to fast after each other. Still figuring out what is really happening.

AlbertHoevenaars, 2021-09-01:
I found out that after an axilite_check a short delay is needed before the next axilite_check (or read/write) starts. I tried to use the insert_delay() command but did not get it working. Could not find that procedure for some reason . Correct libraries are compiled for Questsim (td_vvc_framework_common_methode_pkg.vhd & ti_vvc_framework_support_pkg.vhd). Then I used a "wait for 50 ns;" line after each axilite_check and that worked. Now my simulation ends nicely with a summary of all alerts.

erick166, 2021-09-03:
That's alright, it's good to hear that your testbench works now.

It's difficult to know why you are having problems with the axilite_check() without looking at the testbench. These procedures have been tested even with back-to-back commands so they should work without the need for delays. Perhaps you can look at the UVVM supplementary repository and check the `bitvis_vip_axilite/tb/maintenance_tb/axilite_vvc_simple_tb.vhd` to compare.

AlbertHoevenaars, 2021-09-03:
erikc166, I think that it is not a problem of the uvvm code. It think that it is a failure in the VHDL that I am testing. It is from a project that we took over from another department. They have build a clock crossing inside the AXILite interface. I think, but I have to investigate this further, that the clock crossing is given extra delayed and that ruin's the timing of the valid signals in the AXILite interface. No problem if read or write action are seperated by (in my case) 50 ns. But the UVVM read or write action are close together, which showed this problem in the VHDL. Clockcrossing should not be used in the valid signals but in the register signals after the AXILIte interface.

erick166, 2021-09-03:
Alright, then I guess we can close this issue.

AlbertHoevenaars, 2021-09-07:
Yes, you may close this issue. Thanks for the help

closed 2021-09-07 by erick166