UVVM/UVVM #160: Report procedures cause errors and warnings in *vsim* batch mode

nfritzsc, 2021-08-10:
## Expected Behaviour

I expect to be able to use `report_alert_counters` in *vsim* batch mode without errors and warnings.

## Current Behaviour

Currently each call of `report_alert_counters` causes 2 additional errors and 1 additional warning in the alert summary (see *Failure Logs*). The same issue was observed for `report_global_ctrl`

## Context
We use UVVM Utility Library procedures in a framework launching *vsim* via the command line with `-batch` option with the aim of integrating this in a CI.

## Failure Logs

A test bench ending the simulation with the following lines ...
```vhdl
report_alert_counters(FINAL);                                            
finish;   
```
... results in 2 reported errors and 1 warning caused by `report_alert_counters`:
```bash
# UVVM:      ====================================================================================================================================================================
# UVVM:      *** FINAL SUMMARY OF ALL ALERTS ***
# UVVM:      ====================================================================================================================================================================
# UVVM:                                REGARDED   EXPECTED  IGNORED      Comment?
# UVVM:                NOTE         :      0         0         0         ok
# UVVM:                TB_NOTE      :      0         0         0         ok
# UVVM:                WARNING      :      0         0         0         ok
# UVVM:                TB_WARNING   :      0         0         0         ok
# UVVM:                MANUAL_CHECK :      0         0         0         ok
# UVVM:                ERROR        :      0         0         0         ok
# UVVM:                TB_ERROR     :      0         0         0         ok
# UVVM:                FAILURE      :      0         0         0         ok
# UVVM:                TB_FAILURE   :      0         0         0         ok
# UVVM:      ====================================================================================================================================================================
# UVVM:      >> Simulation SUCCESS: No mismatch between counted and expected serious alerts
# UVVM:      ====================================================================================================================================================================
# UVVM:      
# UVVM:      
# End time: 13:18:00 on Aug 10,2021, Elapsed time: 0:00:02
# Errors: 2, Warnings: 1
```

## Possible Cause

As this issue only occurs in *vsim*'s batch mode (`-batch`), but neither in GUI nor command line (`-c`) mode, it seems to be related to the missing command line.

## Possible Solution

Point out way how to properly use report procedures in *vsim* batch mode

mariuselv, 2021-08-11:
Hi Nick,
Thank you for the report. I will have to test this and investigate the problem.

Br,
Marius

nfritzsc, 2021-08-23:
Hi @mariuselv,

is there any news regarding this issue?

Best regards,
Nick

mariuselv, 2021-08-24:
Hi Nick, 
I have started looking into it, but have not been able to determine what is causing the problem. The errors and warning you get from the simulator at the end of the transcript is somehow triggered by the report procedure. I will notify you when I have found a solution to the problem.

Br,
Marius

UVVM, 2023-08-14:
Hi, This problem no longer exists with any newer simulator, and we could not find the source of the problem in the previous version. Thus, as we expect this is a simulator problem, we close this issue.

BR 
UVVM Team

closed 2023-08-14 by UVVM