UVVM/UVVM #165: Scoreboard input width change

AlbertHoevenaars, 2021-09-07:
I am trying to use the scoreboard but found out that the input of values via the .add_expected function can only be 8 bits width. How can I change this to another value. In my case 16 bit. I found something with "C_SB_SLV_WIDTH", but I am not sure if this is the correct value to change, because all scoreboards will have this width. 

How can I change the width of the expected_value on the scoreboard?

Thanks in advance.

AlbertHoevenaars, 2021-09-13:
Nobody has a solution??????

AlbertHoevenaars, 2021-09-14:
I placed the package definition including the uses statement before the entity of my sequencer. The shared variable I placed between "Architecture" and "begin". I compile all files through a batch file (Questasim), but the package file is not found. Compilation is done based on the files name, so I cannot compile the package slv16_sb_pkg separately. Probably I am doing things wrong, but what?
Are there more examples how this should be done?

package slv16_sb_pkg is new bitvis_vip_scoreboard.generic_sb_pkg 
    generic map(t_element => std_logic_vector(15 downto 0),
    element_match => std_match,
    to_string_element => to_string,
    sb_config_default => C_SB_CONFIG_DEFAULT);

use slv16_sb_pkg.all;
--shared variable slv16_sb : slv16_sb_pkg.t_generic_sb;

ENTITY axi4ic_tb_i2f_stim IS
  PORT( 
    clk            : IN     std_logic;
    rst            : IN     std_logic;
    WaitOut        : IN     std_logic;
    ImageIn        : OUT    tFullInternalFormat4;
    WaitIn         : OUT    std_logic
  );

-- Declarations

END ENTITY axi4ic_tb_i2f_stim ;

ARCHITECTURE behave OF axi4ic_tb_i2f_stim IS

--shared variable TB_SB : my_tb_sb_pkg.t_generic_sb;
shared variable slv16_sb : slv16_sb_pkg.t_generic_sb;

mariuselv, 2021-09-14:
Hi,
You can have a look at the maintenance TB at https://github.com/UVVM/UVVM_SUPPLEMENTARY/blob/68a7140efd2943dd8ac1cb7e4ae5227ba87d3caa/bitvis_vip_scoreboard/tb/maintenance_tb/generic_sb_slv_tb.vhd

This TB is setting up a SB as slv 8 bit wide. 

Br. 
Marius

AlbertHoevenaars, 2021-09-14:
Many thanks. This example solved my problem. Complilation by Questasim is Ok and testbench is running now.

closed 2021-09-14 by mariuselv