UVVM/UVVM #167: bitvis_uart ghdl simulation script broken?

lukipedio, 2021-10-02:
Just tried to run:
`source ghdl_compile_all_and_simulate.sh`

on master branch (commit SHA 3e85c675) and it fails.

mariuselv, 2021-10-02:
Hi,
It's been a while since I tested the GHDL script. Which GHDL version and UVVM version are you running?

Br. 
Marius

lukipedio, 2021-10-02:
I am using testing with GHDL v0.37 + current UVVM master branch.

----- Messaggio originale -----

https://github.com/UVVM/UVVM/issues/167#issuecomment-932757585
 CAEN S.p.A.   Luca ColombiniFrontend Division - Senior Design Engineerl.colombini&#64;caen.it Tel.: &#43;39 0584 388 398Fax: &#43;39 0584 388 959 CAEN S.p.A.via Vetraia,11 • 55049 • Viareggio (LU) • ItalyCod.Fisc. e p.IVA IT00864500467https://www.caen.it AVVISO DI RISERVATEZZA. Le informazioni contenute in questo messaggio di posta elettronica e/o files allegati, sono da considerarsi strettamente riservati. Il loro utilizzo è consentito esclusivamente al destinatario del messaggio, per le finalità indicate nello stesso. Costituisce violazione ai principi dettati dalla normativa nazionale ed europea in materia di Trattamento dei Dati Personali: trattenere il messaggio stesso oltre il tempo necessario, divulgarlo anche in parte, distribuirlo ad altri soggetti, copiarlo o utilizzarlo per finalità diverse. Ricordiamo che in ogni momento potrete richiederci l’informativa per la privacy e la sospensione da parte nostra dell&#39;impiego dei vostri dati, ad esclusione delle comunicazioni effettuate in esecuzione di obblighi di legge. Qualora riceveste questo messaggio senza esserne il destinatario Vi preghiamo cortesemente di darcene notizia via e-mail e di procedere alla distruzione del messaggio stesso, cancellandolo dal Vostro sistema. Grazie.The information contained in this e-mail and/or attached files are to be considered strictly confidential. Their use is allowed exclusively to the recipient of the message, for the indicated purposes. It is a violation of the principles dictated by national and European legislation on the processing of personal data: retain the message after the necessary time, disclose any of its part, distribute it to other subjects, copy it or use it for different purposes. We remind you that at any time you can request privacy information and the suspension from your data use, excluding communications made in compliance with legal obligations. If you received this message without being the recipient, we kindly ask you to inform us via e-mail and delete it from your system. Thank you.

mariuselv, 2021-10-02:
Hi,
It seems to be failing due to an accepted pull-request in the ti_protected_types_pkg on August 16th. I will investigate a bit more to find a solution to the problem, in the meanwhile I suggest that you use version 2021.05.26.

Br,
Marius