UVVM/UVVM #171: Scoreboard input channels

Albert-Hoevenaars, 2021-12-16:
In my VHDL I use a number of UVVM blocks. One of them is the scoreboard. Both DUT and model send values to the scoreboard, but sometimes my model is faster and sometimes my DUT is faster. This gives me the error message that there is no value to check. Swapping expected and checked does not help. But I believed that there is only one buffer in the scoreboard. I would be nice if there would be two buffers in the scoreboard, on each channel one. When both buffers hold data a compare could be done.
Is there at this moment a solution for this problem?

Best Regards,

Albert

EspenTallaksen, 2021-12-16:
Hi Albert,
The system assumes that the model is behavioural and is generating the expected results in zero time, and that the DUT spends time - or at least more delta-cycles. That will of course not be the case if you have RTL in your model, in which case the probability of racing is higher. What is your case?
The problem with your proposed solution is that for several testbenches we need to use the entry of data into the scoreboard as a trigger to compare actual vs expected, and if we don't trigger on the actual data then there could for instance be a problem with matching out of order data and to synchronize after receiving garbage data. 
If you use a dedicated stand-alone Scoreboard, and you cannot know whether the expected or actual data arrives first, I therefore recommend to delay your actual data (i.e. delay the writing of the actual data into the scoreboard), either by a fixed and safe delay, or by waiting for the expected data to be written.  There is currently no built-in support for this.
Your proposed solution would probably work if there is a one to one relation between actual and expected.  We could evaluate whether to make a switch for this in the Scoreboard configuration?

Best regards
Espen

UVVM, 2022-09-28:
Hi,
We have added this request for internal discussions.

Br,
UVVM Team

closed 2022-09-28 by UVVM