UVVM/UVVM #172: Questasim 2019.1: vsim -c -do compile_all.do returns File exists. (errno = EEXIST)

fransschreuder, 2022-02-02:
Hello,
When I try to compile UVVM with questasim 2019.1, I always run into this compile_all.do script returning with an error.
```
# eval vcom  -quiet -suppress 1346,1236,1090 -2008 -work uvvm_util  /localstore/et/franss/felix/firmware5/simulation/UVVM/script/../uvvm_util/script/../src/uvvm_util_context.vhd
Unknown simulator. Attempting to use Modelsim commands.
# 
# 
# === Re-gen lib and compile uvvm_vvc_framework source
# 
# Source path: /localstore/et/franss/felix/firmware5/simulation/UVVM/script/../uvvm_vvc_framework
# Target path: /localstore/et/franss/felix/firmware5/simulation/UVVM/script/../uvvm_vvc_framework/sim
# ** Error: (vlib-35) Failed to create directory "/localstore/et/franss/felix/firmware5/simulation/UVVM/script/../uvvm_vvc_framework/sim/uvvm_vvc_framework".
# File exists. (errno = EEXIST)
# Executing ONERROR command at macro ./compile_all.do line 93
```

When I run it again, it get's a step further, and finally when I put the script in a bash while loop. grepping for ONERROR, it succeeds. Is there something I am doing wrong, or an incompatibility with questasim (on Linux)?

UVVM, 2022-06-01:
Hi,
Sorry for the late response. We will have a check what is causing this issue.

Br,
Marius

fransschreuder, 2023-03-30:
I run the command from UVVM/script. This is what I currently do to work around the issue:
```
cd ../UVVM/script/
while vsim -c -do ./compile_all.do -do quit|grep EEXIST; 
do
    echo compiling uvvm
done

```

fransschreuder, 2023-03-30:
This is the output I get, and finally it ends in a successful compilation of the library:
```
# File exists. (errno = EEXIST)
compiling uvvm
# File exists. (errno = EEXIST)
compiling uvvm
# File exists. (errno = EEXIST)
compiling uvvm
# File exists. (errno = EEXIST)
compiling uvvm
# File exists. (errno = EEXIST)
compiling uvvm
# File exists. (errno = EEXIST)
compiling uvvm
# File exists. (errno = EEXIST)
compiling uvvm
# File exists. (errno = EEXIST)
compiling uvvm
# File exists. (errno = EEXIST)
compiling uvvm
# File exists. (errno = EEXIST)
compiling uvvm
# File exists. (errno = EEXIST)
compiling uvvm
# File exists. (errno = EEXIST)
compiling uvvm
# File exists. (errno = EEXIST)
compiling uvvm
# File exists. (errno = EEXIST)
compiling uvvm
# File exists. (errno = EEXIST)
compiling uvvm
# File exists. (errno = EEXIST)
compiling uvvm
# File exists. (errno = EEXIST)
compiling uvvm
# File exists. (errno = EEXIST)
compiling uvvm
# File exists. (errno = EEXIST)
compiling uvvm
# File exists. (errno = EEXIST)
compiling uvvm
# File exists. (errno = EEXIST)
compiling uvvm

```

fransschreuder, 2025-10-21:
Hi, the compile_all.sh script doesn't work for me either. I think I found the issue:
file_mkdir should only make $target_path, not $target_path/$lib_name (as also checked within the if statement:
With this fix, my compiles are going smoothly.

```
diff --git a/script/compile_src.do b/script/compile_src.do
index 2a2ef47..10183b3 100644
--- a/script/compile_src.do
+++ b/script/compile_src.do
@@ -82,7 +82,7 @@ if {[file exists $target_path/$lib_name]} {
   file delete -force $target_path/$lib_name
 }
 if {![file exists $target_path]} {
-  file mkdir $target_path/$lib_name
+  file mkdir $target_path
 }
 
 quietly vlib $target_path/$lib_name
@@ -124,4 +124,4 @@ foreach item $file_data {
     eval vcom  $compdirectives  $source_path/script/$item
   }
   incr idx 1
```

UVVM, 2025-10-22:
Alright, I see it gives the same result in my setup since `quietly vlib $target_path/$lib_name` creates the library directory anyway.
We'll add this fix in the next release.