UVVM/UVVM #173: Await stable value function

andrefi-eml, 2022-02-15:
Hi!

One feature that I would like to have in the UVVM utility library is a modified version of await_stable where I can specify which value I want to wait for.
What I want to do is to check that a signal has a given value, and that the signal keeps this value over a time period.

Today I often write this over several lines of code:
check_value(X, '0', "Check that X is zero")
timestamp = now;
wait for 1 ms;
check_value(X, '0', "Check that X is still zero")
check_stable(X, (now-timestamp), "Check that X was stable over the given time")

It would be nice to instead have something like:
await_stable_value(X, '0', 0 ns, 1 ms, "Checking that X is zero and remains zero for 1 ms");

Even better, it would be a good idea to have a non-blocking version that can run in the background:
await_stable_value(X, '0', 0 ns, 1 ms, NON_BLOCKING, "Checking that X is zero and remains zero for 1 ms while I do something else");

Thanks and best regards,
Andre

UVVM, 2022-09-28:
Hi Andre,

Could you create a pull-request for this feature?

Br,
UVVM Team

EspenTallaksen, 2022-10-04:
Hi Andre, 
- and thanks for the pull request :-)
I understand your point about the name, but 'await_*' is used to indicate that the procedure is time consuming and that you (potenitally) have to wait for it to finish, so I think the name should start with 'await_', or otherwise we break the "rule".

andrefi-eml, 2022-10-04:
I have updated the pull request and replaced the name check_stable with await_stable_value.

closed 2022-10-05 by andrefi-eml