UVVM/UVVM #174: RGMII BFM: Ability to clock data on rising edge only

dbemlogic, 2022-02-21:
According to Wikipedia (with reference) the RGMII clocks data on both edges for 1 Gbps operation, but only on rising edge for 100 Mbps and 10 Mbps. 

I would like support for clocking data on rising edge only for this BFM.

https://en.wikipedia.org/wiki/Media-independent_interface

"Data is clocked on rising and falling edges for 1000 Mbit/s, and on rising edges only for 10/100 Mbit/s.[[7]] The RX_CTL signal carries RXDV (data valid) on the rising edge, and (RXDV xor RXER) on the falling edge. The TX_CTL signal likewise carries TXEN on rising edge and (TXEN xor TXER) on the falling edge. This is the case for both 1000 Mbit/s and 10/100 Mbit/s.[[8]]"

UVVM, 2022-02-21:
Hi Daniel, 
Could you provide a pull request for this?
We hope that users (like your customer) are willing to invest some time in improving UVVM and feeding this back to the community, as we assume they all save a lot of work/time/cost by using UVVM, and we all benefit from new functionality from the community.

-- Espen

dbemlogic, 2022-02-22:
I will provide a pull request. :)

UVVM, 2022-02-22:
Great. Thanks :-)

dbemlogic, 2023-04-12:
The answer to that is that I completely forgot about it.... I will look at it this week.

Mvh,

Daniel Blomkvist
Senior FPGA Designer
EmLogic AS

Mob.: +47 91 58 97 56
***@***.***
www.emlogic.no

Hi Daniel,
How is it going with the pull request?

Marcos-4zeved0, 2023-09-07:
@UVVM There's any parameter that could change clock polarity into UVVM standard bfm procedures ? If doesn't, I would glad help, but I'll need support to understand faster:
- Clock flow into your packages
- Where bfm procedures do clk management

Could you please do a brief explanation ? Hope we can help each others :)

UVVM, 2024-08-20:
> @UVVM There's any parameter that could change clock polarity into UVVM standard bfm procedures ? If doesn't, I would glad help, but I'll need support to understand faster:
> 
> * Clock flow into your packages
> * Where bfm procedures do clk management
> 
> Could you please do a brief explanation ? Hope we can help each others :)
> 
> edit: need one feature too: A parameter in check procedures to wait for X cycles before gives a error, cause my processes have to wait for 1 cycle after read to awnser.

Sorry for the long delay in answering. The BFM procedures receive the clock signal as an input parameter and they sample or transmit the data at the corresponding clock edge depending on the protocol. I don't fully understand why you would want to change the clock polarity in the BFM procedures, but if this is still an issue you would like to discuss could you create a separate Github issue?

As for the feature, perhaps you could use instead await_value() with a max_time equal to one clock cycle?

closed 2024-09-20 by UVVM