UVVM/UVVM #175: Message IDs

ashokmehta72, 2022-02-23:
Hello, 

Brand new to UVVM. Not sure if I understand the idea behind Message IDs.

1.  ID_FINISH_OR_STOP. What does this ID do? If I use it in log(ID_FINISH_OR_STOP,...); does it actually Stop or Finish the simulation or simply display a message 
    UVVM : ID_FINISH_OR_STOP   3000.00 ns TB_Seq ....

   Also, the examples I see on net always show the scope as TB_seq. What is TB_seq? Name of the entity as the scope? Or the architecture? Or the process?

2. ID_BFM. Again, what does this accomplish? Does it have to be used in a BFM VVC? What if I use it outside of a BFM? What's the main purpose of using it?

3. Similarly, don't understand what ID_UVVM_SEND_CMD do? Google search tells me "shows commands being sent to queues".  Could you please further elaborate on how this message ID figures out which commands are being sent to queues?

Thanks. Like the overall flow of UVVM. Similar to SystemVerilog UVM.

UVVM, 2022-02-24:
Hi,
These IDs are for verbosity control only.
Please check out the provided powerpoints. They should give you the explanation you need on this and other beginners' issues.

-- Espen

ashokmehta72, 2022-02-24:
Thanks Espen for the link to documents. I was aware of some of these documents but not all. Better understand IDs meaning.

ashokmehta72, 2022-03-30:
> Thanks Espen for the link to documents. I was aware of some of these documents but not all. Better understand IDs meaning.

I went through pretty much all the documents. But still don't understand what is "TB Seq" that I see coming out in log messages. For example, I have the following line in my code:

log(ID_LOG_HDR, "Starting simulation of TB for counter using VVCs", C_SCOPE);

The log message I get is:
# UVVM: ID_LOG_HDR                         0.0 ns  TB seq.                        Starting simulation of TB for counter using VVCs

What is "TB seq" in this header? Has this something to do with C_SCOPE? As I understand C_SCOPE always refers to the scope in which the log procedure is called. Maybe you could explain C_SCOPE.

My Entity/arch. name is counter_vvc_tb/functional. There is no "TB seq" as the name anywhere in my design.

Thanks.

EspenTallaksen, 2022-03-31:
Hi,
Scope is: From where did this message originate, i.e. from where in the testbench hierarchy.
The main or central test sequencer has been denoted 'Testbench sequencer' - hence  'TB seq' in short.
If you generate messages from a support process or entity (e.g. av verification component) you would want that to be the scope.
- Espen

ashokmehta72, 2022-03-31:
Hi Espen, Thanks very much for explaining. My understanding of scope is in line with your explanation.

Still a question on TB_Seq. My log message is from a top level entity counter_vvc_tb with architecture name 'functional'. I was hoping to see the scope name as 'functional' or counter_vvc_tb/functional. How does the main or central test sequencer come into picture here? Or is it some sore of default scope for top level testbench? 

Hope my question makes sense.

Thanks.

EspenTallaksen, 2022-04-25:
Hi,
Sorry, this question disappeared during a long Easter vacation.
To get the scope you want in your testbench, you need to add that string as the scope parameter to all your local log messages originating from that scope - preferably as a constant to assure consistency.
-- Espen

closed 2022-09-28 by UVVM