UVVM/UVVM #178: uvvm_util: rand_pkg / func_cov_pkg causes Modelsim error

ThiemoClausen, 2022-04-20:
Hi,

### Environment:
uvvm_util versions: 2.18.0 / 2.18.1
Modelsim Version: 2019.1 PE / 2019.1 DE 64 Bit
Modelsim Code Coverage: "s" (statement coverage enabled)

### Observation:
While starting a UVVM based simulation with code coverage collection enabled, Modelsim gives me the the following errors:

 ```
#    Time: 0 fs  Iteration: 0  Region: /methods_pkg File: C:/<somePath>/src/verif_ip/3rd_party/uvvm_util/2.18.1/src/methods_pkg.vhd
# Loading uvvm_util.bfm_common_pkg(body)
# Loading uvvm_util.generic_queue_pkg(body)
# Loading uvvm_util.rand_pkg(body)
# ** Error: Internal Error - process_push_cov_static_scope_int: target static scope 218 not found
# . Please contact customer support for further assistance.
# ** Error: Internal Error - process_push_cov_static_scope_int: target static scope 219 not found
# . Please contact customer support for further assistance.
# ** Error: Internal Error - process_push_cov_static_scope_int: target static scope 220 not found
# . Please contact customer support for further assistance.
# ** Error: Internal Error - process_push_cov_static_scope_int: target static scope 221 not found
# . Please contact customer support for further assistance.
# ** Error: Internal Error - process_push_cov_static_scope_int: target static scope 222 not found
# . Please contact customer support for further assistance.
# ** Error: Internal Error - process_push_cov_static_scope_int: target static scope 223 not found
# . Please contact customer support for further assistance.
# ** Error: Internal Error - process_push_cov_static_scope_int: target static scope 224 not found
# . Please contact customer support for further assistance.
# ** Error: Internal Error - process_push_cov_static_scope_int: target static scope 225 not found
# . Please contact customer support for further assistance.
# ** Error: Internal Error - process_push_cov_static_scope_int: target static scope 226 not found
# . Please contact customer support for further assistance.
# ** Error: Internal Error - process_push_cov_static_scope_int: target static scope 227 not found
# . Please contact customer support for further assistance.
# ** Error: Internal Error - process_push_cov_static_scope_int: target static scope 228 not found
# . Please contact customer support for further assistance.
# ** Error: Internal Error - process_push_cov_static_scope_int: target static scope 229 not found
# . Please contact customer support for further assistance.
# ** Error: Internal Error - process_push_cov_static_scope_int: target static scope 230 not found
# . Please contact customer support for further assistance.
# Loading uvvm_util.func_cov_pkg(body)
# ** Error: Internal Error - process_push_cov_static_scope_int: target static scope 103 not found
# . Please contact customer support for further assistance.
# ** Error: Internal Error - process_push_cov_static_scope_int: target static scope 105 not found
# . Please contact customer support for further assistance.
# ** Error: Internal Error - process_push_cov_static_scope_int: target static scope 104 not found
# . Please contact customer support for further assistance.
# ** Error: Internal Error - process_push_cov_static_scope_int: target static scope 106 not found
# . Please contact customer support for further assistance.
```

If I remove the rand_pkg and the func_cov_pkg from the uvvm_util by removing it from compile order list and removing the related entries in the uvvm_util_context.vhd, the errors are gone!

Any ideas on what is going on there?

kind regards
Thiemo

ThiemoClausen, 2022-04-21:
Hi Espen,

code coverage collection is enabled for all files, including test bench and all the libs (UVVM included).
As far as I know, modelsim cannot selectively collect this data and one can only exclude files or design units after the simulation with coverage collection enabled is done.

Regards
Thiemo

EspenTallaksen, 2022-04-22:
Hi Thiemo,

That is really strange. Are you sure?
I haven't tried this lately, but that was not how I used to do this, and it doesn't quite make sense. The only thing you are interested in, is coverage on your design, which also means that including coverage on your testbench++ will slow down your simulation (and give you no information of interest)
It seems this is still the way it works:
Check out https://users.ece.cmu.edu/~kbiswas/se_man.pdf page 334

Regards
Espen

ThiemoClausen, 2022-04-22:
Hi Espen,

thanks, you are correct, I double checked, collective coverage collection should work.
I will modify my compile scripts accordingly and see if this works around the issue.

Nevertheless, Modelsim should not throw errors while dealing with the rand_pkg and the func_cov_pkg with coverage enabled.
Not sure if this is a Modelsim or a UVVM issue though, I had not the time to dig deeper into the rand_pkg and the func_cov_pkg sources yet.

Regards,
Thiemo

EspenTallaksen, 2022-04-22:
Hi Thiemo,

I agree that it is still strange that you got these errors.
It is difficult to see how this could be caused by UVVM without the complete testbench, and it could be hard to debug if it is a Modelsim problem. 
Let's hope your TB runs ok with your changes.

Regards
Espen

ThiemoClausen, 2022-04-22:
Hi,

I can confirm that the TB runs fine when coverage options are not set for TB files.

Regards
Thiemo

closed 2022-04-22 by EspenTallaksen