UVVM/UVVM #183: Difference between alert(TB_WARNING, "..."); and tb_warning("...");

ashokmehta72, 2022-05-04:
Both the alert(TB_WARNING,...) and procedure tb_warning("..."); seems to give identical log messages. When would you use one over the other? The only difference I see is that with alert you can control severity level, output to alert file, etc. But why do you have two different ways to output the same message?

EspenTallaksen, 2022-05-05:
Normally you would not use any of those procedures directly in your testbench, but rather checkers, awaits, etc. - where those procedures are included. There is no difference between the alert(tb_warning, ...) and tb_warning(...). The reason for having both are different historical needs.
-- Espen

EspenTallaksen, 2022-05-24:
TB_* is meant to be used when you know for sure that this alert is a pure testbench problem. For instance if you try to write a 16-bit value onto an 8-bit signal in a BFM. This could be useful information for the user. If you don't like this feature, you could just avoid using them in your own testbench.
-- Espen

closed 2022-06-01 by mariuselv